arendst/Tasmota #1765: no access to Web with Hight round trip delay.

storm66-ag, 2018-02-01:
Hello. Could any body try to remote access to sonoff  with ping delay more that 100-200ms. 
I set port forwarding, and now there is not access from internet with hight latency, but from internet with good latency its working good. Good latency is about 1 - 40ms. 
I believe this is Tasmota SW  WEB server settings???

smadds, 2018-02-01:
If you are using MQTT, check that it is connected to your mqtthost properly as this can have long timeouts.

If you have installed an mDNS service on the mqtthost you can go to the sonoff's console page and issue the command `mqtthost 0`. This will clear the mqtthost settings and cause the device to search for a new host.

storm66-ag, 2018-02-01:
The MQTT disabled on my sonoff module. I need web page. Its really so I sad,  when rtt is going up (upper that 60ms)  - no access to module, than rtt is going to low (about 20-30 ms) its working again.

smadds, 2018-02-01:
Is the Wifi signal OK? You can see the strength on the Information page when you can access the web page

storm66-ag, 2018-02-01:
Scheme is: 
sonoff<->wifi<->router<->LTE<->Internet-.......

storm66-ag, 2018-02-01:
Signal is ok, but I see that latency is changing periodically (upper/lower).

gajotnt, 2018-02-01:
I have a sonoff thats im my yard and barely gets wifi signal, mqtt works well, but web interface only sometimes...
You should consider using MQTT

sonoffheld, 2018-02-02:
i think is the same problem in #1711

storm66-ag, 2018-02-05:
Mqtt is working good, no problem. But I need web panel view. I don't want to use the external cloud for getting mqtt value.

closed 2018-06-20 by stale[bot]