arendst/Tasmota #1766: LED Behaviour  (Bug or Design ) Wemos device

shutty1, 2018-02-01:
I have managed to flash a INSMA 3 way wifi touch switch which is using an esp-12e.
The pinouts I have discovered are below :

I have used the wemos device as that seemed the best sited following another post i read.

I am not sure how much difference using a different device is other than of course the pintouts, eg is there any special coding for the different device types or is it all generic.

The switch initialises ok and the leds all set correctly.
But when i press the switch manually or use the web interface to toggle the switch, all three switches only adjust the top switch led.
(in other words led 2 and led 3 are not responding , it is using led1 instead.)

Will i need to look at the code to find this ?,  A bug or simply designed for the sonoff touch which may behave differently etc ?  any guidance would be appreciated.

"WeMos D1 mini",   // WeMos and NodeMCU hardware (ESP8266)
     GPIO_LED4,        // GPIO00     (green led, not sure of purpose)
     GPIO_USER,        // GPIO01
     GPIO_LED3,        // GPIO02    (bottom switch blue off, red on led)
     GPIO_KEY2,        // GPIO03 
     GPIO_REL2,        // GPIO04 
     GPIO_KEY3,        // GPIO05 
     0, 0, 0, 0, 0, 0,       // 6,7,8,9,10,11  (cant use)
     GPIO_KEY1,        // GPIO12    
     GPIO_REL1,        // GPIO13 
     GPIO_LED2,        // GPIO14     (middle switch blue off, red on led)
     GPIO_REL3,        // GPIO15 
     GPIO_LED1,        // GPIO16      (top switch blue off, red on led)
     GPIO_ADC0         // ADC0   A0 Analog input
  },

reloxx13, 2018-02-02:
Did u set the leds on the web interface?...

shutty1, 2018-02-10:
I cant see any options to set the leds on the web interface other than the assignments i made above... 
GPIO_LED1 , GPIO_LED2 ETC...  i initialy used the configuration module in the web-interface  to identify the correct pins etc and i know they are correct now.  But in real use they all default to the 1st led, now that i have looked at the http commands available , it does look like there is only support for one led, it does not have the option like power does <power><x> where you can specify which relay/switch you want to work with.  so the conclusion i am reaching is that multiple led control for a 2 or 3 way switch has not been built into the device driver at this point. I can even see some reference in the code that appears to hardcode to led1 only. (but i have only had a brief look at this point)