arendst/Tasmota #1767: Mqtt connection

Caswell1000, 2018-02-01:
Please assist, I flashed tasmota firmware on sonoff basic, configured mqtt with own user, client, and password. I then installed mqtt dashboard and entered same user, client name and password. I can't seem to connect to the device using the mqtt app.

Caswell1000, 2018-02-01:
Am using a cloud broker broker.mqttdashboard.com where I entered my sonoff mqtt details, topic is sonoff, and I publish "cmnd/sonoff/power toggle" to no avail.

gnkarn, 2018-02-03:
what you mean by mqtt dashboard ?
in order to test what you just did, i think first , you should login into your sonoff , by using your sonoff IP .
from there , verify the mutt configuration , and finally test , by using the  power on off command from the Console .

Caswell1000, 2018-02-06:
Thanks, I tested the sonof on the sonoff ip, all commands work, I installed an app called mqtt dashboard, when I type in my topic/cmnd/power on the app it works and gives me the status on the sonoff ip console, however, cmnd/power on or cmnd/power 1 or cmnd/power 2 return an unknown command like this) 
......... Stat/RESULT = {"Command":"Unknown"}. This is how it's given in the example so I'm confused why only cmnd/power works but the rest fail, it seems the payload is not recognised.

Caswell1000, 2018-02-08:
I'm on version 5.11.1 any power commands with payloads like "/cmnd/Testchip/power on"  return a "command unknown"   result on the console. A power command without a payload "/cmnd/Testchip/power" returns the status of "OFF" in the console. I am currently connected via USB only. On the console,  "power on" yet urns the desired result. Why is mqtt not accepting power commands with payloads like "on" or "1"

Caswell1000, 2018-02-08:
I am closing this issue because I have clarified the question in another issue #1827

closed 2018-02-08 by Caswell1000