arendst/Tasmota #1771: Need some help with MQTT

larrybell, 2018-02-02:
I am using a NODEMCU with Tasmota v5.10 installed.
I changed the Module type to 18 WeMos D1 mini
D0 GPIO16 is set to 48 Led1

I am trying to understand how for example I would turn on and off the onboard LED via MQTT.
I have am running Mosquitto and Node-Red. 

Your help would be greatly appreciated.

curzon01, 2018-02-03:
If you want control the LED independently from program logic use PWMx for definition, not LEDx:

E. g. blue onboard LED - connected to GPIO2 (D4)

Configure in Module Config D4 GPIO2: PWMi1

Control LED now with command 
Pwm1 0 - off
Pwm1 1023 - on

Other values between 0 and 1023 will dim the LED - see also [Management Commands](https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#management)

larrybell, 2018-02-04:
Cool that works great. Lets say I want to use the NodeMCU as a WiFi button. I have been playing with this for a while but not had any luck with getting it to work. FYI I am a beginner at this.

curzon01, 2018-02-04:
if this issue is no longer an issue for you pls close it

closed 2018-02-04 by larrybell