arendst/Tasmota #1772: Cant retrieve data. 5.11.1

GAZ082, 2018-02-03:
Hi there. I have configured my full topic as:

`pola/living/%topic%/%prefix%/`

i wanna get the status, so i issue the command:

`pola/living/device/Status `

with no payload and then i subscribe to:

`pola/living/device/stat/+`

i get nothing.

i can issue commands with no reply such as power:

`pola/living/device/POWER`

but for some reason can get the Status work! Nor any other command which i'd get a reply. I believe i'm messing things with the topic i subscribed to get the reply,

I'm working with Node Red.

Tips? Thanks!

curzon01, 2018-02-03:
commands must be publish using prefix `cmnd`:

`pola/living/device/cmnd/Status`

`pola/living/device/cmnd/POWER`

Please read again [MQTT Topic definition](https://github.com/arendst/Sonoff-Tasmota/wiki/MQTT-Features#mqtt-topic-definition) carefully

GAZ082, 2018-02-03:
Hi @curzon01, thanks for noting that. The truth is it was late in the night and was tired, but i DO use the prefixes when issuing the commands, the thing is the subscription to listen the data:

`pola/living/device/stat/+`

gets me nothing.

here is my NR flow just to get the example:

`[{"id":"769e7e5f.99b9f","type":"inject","z":"f53db186.9b7d3","name":"","topic":"pola/living/ventilador/cmnd/Status/","payload":"","payloadType":"num","repeat":"","crontab":"","once":false,"x":110,"y":90,"wires":[["34e9718a.26891e"]]}]`

curzon01, 2018-02-03:
What does the Web Console on Sonoff device display when you publish a "cmnd" topic? Is there any logging info on Sonoff when you publish something via NR? 

If not, do not search on Tasmota for issue, check your MQTT server/authentication and topic permissons first.
Most time when MQTT will not respond as expected it is a mqtt server permission issue - either the user/pw combination is not allowed to publish such topics or subscribe it.

GAZ082, 2018-02-03:
Okey now. That's for the idea of working with the console. Found where the response was being sent, it did not follow the full topic. will check with the NR guys. Thanks.

closed 2018-02-03 by GAZ082