arendst/Tasmota #1775: MQTT IR-received --> MOD FOR Generating the Mqtt command -  (QUESTION)

gnkarn, 2018-02-03:
i have a device that generates codes I'm reading by using the IR library .

By looking at the IRreceiveCheck() inside the StateLoop() , it seams to me that i need to put the code to generate the mqtt publish message there ( in my case for Home assistant) .

I see that there is an option inside that function to generate the Mqtt message for Domoticz , 
my question , is if my understanding is correct, or there is a better suggestion on where  made the modifications so  to keep the deviations with the original tasmota as minimum as possible, thank you

arendst, 2018-02-03:
You're correct.

If specific for Home Assistant put it between a new define USE_HOME_ASSISTANT / endif just below the USE_DOMOTICZ define.

gnkarn, 2018-02-04:
Thank you for your great job and support ¡

gnkarn, 2018-05-24:
used , to send the MPX decoded data to Home assistant after decoding the signals from the alarm pannel.

closed 2018-05-24 by gnkarn