arendst/Tasmota #1776: Adafruit io not supported?

idandylux, 2018-02-03:
Hello, 
whole Saturday I was trying to connect Tasmota to Mqtt io.adafruit.
Result: was not working.
My question: Is adafruit Mqtt supportet by Tasmota?
Thanks a lot.
andy
luxemburg

closed 2018-06-18 by stale[bot]

muuuh, 2018-12-23:
Same problem here.

One issue seems to be that the **mqtt password is limited to 32 characters** in the Tasmota firmware (at least what I see on "http://.../mq").

According to here: https://learn.adafruit.com/welcome-to-adafruit-io/mqtt-api-documentation-2#connection-details-16-2 and the website "YOUR AIO KEY" in the https://io.adafruit.com/ Dashboard, the MQTT password is fixed to 40 characters

st2000, 2019-08-29:
Another possible problem with using Tasmota & io.adafruit.com.  Tasmota includes all capital words like SENSOR in their topic names.  I don't think this is something that can be changed through normal settings.  The problems is that io.adafruit.com only allows lower case letters, numbers & dashes in their topic names.  So far I have not found a way around this.