arendst/Tasmota #1777: How to set TimeZone to 5.5

sdhawade, 2018-02-03:
Hi I live in India and am trying out this great piece of software, however there seems to be one oversight that the Timezone variable is an int. Can you let me know how to setup the same for GMT+5.5 ?

arendst, 2018-02-03:
You'll have to re-compile the code with a change in user_config.h (or better user_config_override.h) regarding your location using the defines below.
```
// -- Time - Start Daylight Saving Time and timezone offset from UTC in minutes
#define TIME_DST               North, Last, Sun, Mar, 2, +120  // Northern Hemisphere, Last sunday in march at 02:00 +120 minutes

// -- Time - Start Standard Time and timezone offset from UTC in minutes
#define TIME_STD               North, Last, Sun, Oct, 3, +60   // Northern Hemisphere, Last sunday in october 02:00 +60 minutes
```
Once loaded you execute command ``timezone 99`` and that's it.

sdhawade, 2018-02-03:
Thanks ! That was quick. Will try it out and let you know.

sdhawade, 2018-02-05:
Worked like a charm ! Thanks a lot !

closed 2018-02-05 by sdhawade

earth08, 2018-03-27:
dear sdhawade,
how did u make it?
Please guide

sdhawade, 2018-03-27:
As mentioned by @arendst define TIME_DST and TIME_STD as the same and select TZ as 99

aaamitsingh, 2018-04-14:
@sdhawade Please share your ntp server and other settings I'm also facing the same problem. This is my setting
```
// -- Time - Up to three NTP servers in your region
#define NTP_SERVER1            "pool.ntp.org"       // [NtpServer1] Select first NTP server by name or IP address (129.250.35.250) pool.ntp.org
#define NTP_SERVER2            "nl.pool.ntp.org"    // [NtpServer2] Select second NTP server by name or IP address (5.39.184.5) nl.pool.ntp.org
#define NTP_SERVER3            "0.nl.pool.ntp.org"  // [NtpServer3] Select third NTP server by name or IP address (93.94.224.67) 0.nl.pool.ntp.org

#define TIME_DST               North, Last, Sun, Mar, 2, +120
#define TIME_STD               North, Last, Sun, Mar, 2, +120 
#define APP_TIMEZONE           99  
```

earth08, 2018-04-15:
dear dont play with ntp as it is not going to help,
try changing 
#define TIME_DST               North, Last, Sun, Mar, 2, +120  (**# change +120 to +330**)
#define TIME_STD               North, Last, Sun, Oct, 3, +60   (**# change +60 to +330**)
#define APP_TIMEZONE           99 ( this is correct)

asking23, 2018-09-03:
i tried changing TIME_STD_OFFSET +330 but still not effective... what could be the issue ?

ascillato, 2018-09-03:
use the commands in the console

asking23, 2018-09-03:
TimeSTD T +330 like this in console and then press enter right ?

ascillato, 2018-09-03:
All the commands and how to use them are in the wiki at https://github.com/arendst/Sonoff-Tasmota/wiki/Commands (please read the wiki)

For set your time zone **_in last TASMOTA version_**, just type in the console:

```
timezone 99
TimeDST H,W,M,D,h,T
TimeSTD H,W,M,D,h,T
```

where 

Parameter  | Description
-------------| ------------
H | Hemisphere : (0 to 1) 0=Northern, 1=Southern Hemisphere
W | Week : (0 to 4) 0=Last week of the month, 1=First, 2=Second, 3=Third, 4=Fourth
M | Month : (1 to 12) 1=Jan, 2=Feb, ... 12=Dec
D | DayOfWeek : (1 to 7) 1=Sun, 2=Mon, ... 7=Sat
h | Hour : 0-23
T | TimeZone : offset from UTC in MINUTES (-780 to 780) (780min/60min = 13hours)

asking23, 2018-09-03:
For those who are searching for Setting time to +5:30 Indian Standard Time Please use below Commands in Console one by one
TimeSTD 0,0,10,1,2,330
TimeDST 0,0,10,1,2,330
timezone 99

Finally Got it... thanks.

ascillato, 2018-09-03:
great!!!!!

poraschaudhary, 2018-09-30:
> For those who are searching for Setting time to +5:30 Indian Standard Time Please use below Commands in Console one by one
> TimeSTD 0,0,10,1,2,330
> TimeDST 0,0,10,1,2,330
> timezone 99
> 
> Finally Got it... thanks.

Worked like a charm!
I had to change the DNS server to 8.8.8.8 though. Cause if you don't assign a DNS server while installing tasmota it defaults to the gateway of your internet..
ipaddress4 8.8.8.8

barbudor, 2020-04-18:
SOLVED in 
### 6.4.0 20181217
- Add minutes to commands Timezone to allow all possible world timezones

meingraham, 2020-04-18:
... and documented

https://tasmota.github.io/docs/Commands/#timezone
... in hours and  **minutes**

Or, if daylight saving policies apply - https://tasmota.github.io/docs/Commands/#timedst
... timezone (-780..780) offset from UTC in **MINUTES**

Amruth-Smaragdine, 2022-05-27:
"timezone +05:30" command from the console will solve the issue.

sfromis, 2022-05-27:
Yeah, that's what barbudor and meingraham posted about in the two preceding posts.