arendst/Tasmota #1783: PWM command not publishing

netomx, 2018-02-04:
As title. Use PWM, send cmnd/sonoff/PWM 1000

It will not get back the value

curzon01, 2018-02-05:
What you mean is that PWM command returns an empty JSON object:
```
RSL: Group 0, Index 1, Command PWM, Data 1000
MQT: dev/stat/tasmota/phesp03/RESULT = {"PWM":{}}
```

arendst, 2018-02-05:
OK. Got it. When no PWM channel is configured and you execute the PWM command then this is the result.

I will check for any PMW configured and if none I will return 
```
RSL: Group 0, Index 1, Command PWM, Data 1000
MQT: stat/wemos6/RESULT = {"Command":"Unknown"}
```

netomx, 2018-02-05:
I have pwm1 configured, nd still no output.

curzon01, 2018-02-06:
@netomx: Be aware to configure one of the GPIOx as PWM(i), otherwise PWM unconfigured - using cmnd PWMx is not enough to have PWM configured.

And - what is the output of "Status 2"? Do you using 5.11.1i?

netomx, 2018-02-06:
Version: Sonoff-Tasmota 5.11.1 by Theo Arends
GPIO5 = 33 PWM1
I can send commands, and the PWM is working, but it doesn't return the value. 

Example 1 (another sonoff with RELAY):
mosquitto_sub -h IP -u user -P pass -t stat/sonoff1/POWER
ON
OFF

Example 2 (this NodeMCU v3 (nodemcu v2 clone) configured ad wemos d1 mini, only GPIO configured):
mosquitto_sub -h IP -u user -P password -t stat/sonoff2/PWM1 (tried with PWM too)
<no output when sending command>

Status 2:
18:50:55 CMD: STATUS 2
18:50:55 MQT: stat/led1/STATUS2 = {"StatusFWR":{"Version":"5.11.1","BuildDateTime":"2018-01-07T21:51:09","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}

netomx, 2018-02-06:
More exmaples:

`22:10:40 CMD: POWER
22:10:40 MQT: stat/cuarto/RESULT = {"POWER":"OFF"}
22:10:40 MQT: stat/cuarto/POWER = OFF`

`22:11:12 CMD: PWM 1000
22:11:12 MQT: stat/led1/RESULT = {"PWM":{"PWM1":1000}}`

arendst, 2018-02-06:
@netomx I see you are still using the old version 5.11.1 while I reported above that your problem is fixed in 5.11.1i.

Install that one and report your findings agian.

netomx, 2018-02-06:
Forgot to say, I upgraded it before the "more examples" post:

`22:08:39 CMD: status 2
22:08:39 MQT: stat/led1/STATUS2 = {"StatusFWR":{"Version":"5.11.1i","BuildDateTime":"2018-02-06T14:12:30","Boot":31,"Core":"2_4_0","SDK":"2.1.0(deb1901)"}}`

arendst, 2018-02-06:
Right. So what's the issue here. I see nothing wrong in your latest entries.

netomx, 2018-02-06:
I will try to be as simple and clear as possible, language is barrier here:

When I send a command to the device, it replies with 2 lines:

RESULT
COMMAND (for example, POWER)

but if I send PWM command, it just replies with:

RESULT

That is my issue

arendst, 2018-02-06:
As designed. Only power command returns both JSON and for legacy reasons also plain results.

To switch naming of result to command you can select one of the SetOptions. See wiki.

netomx, 2018-02-06:
F*ck. Thanks, pal. I will close it as a unrelated bug was fixed and "my bug" was not.

closed 2018-02-06 by netomx