arendst/Tasmota #1784: Override HA autodiscovery switch => light

pascalsaul, 2018-02-04:
`ha/**switch**/basic_woonkamer_bank/config {"name":"basic_woonkamer_bank","command_topic":"cmnd/basic_woonkamer_bank/POWER","state_topic":"stat/basic_woonkamer_bank/RESULT","value_template":"{{value_json.POWER}}","payload_off":"0","payload_on":"1","availability_topic":"tele/basic_woonkamer_bank/LWT","payload_available":"Online","payload_not_available":"Offline"}`

Is it possbile to create an option to override 'switch' to 'light' with the autodiscovery so it will be recognized in the 'Light' group and not in 'Switch' group?

emontnemery, 2018-02-18:
'Light' will be used for pwm dimmer when setoption15 is 1, otherwise 'switch'

francescomapelli, 2018-06-28:
+1 for autodiscovery with light instead of switch. 
I have dozen of sonoff in the walls of my house and they control the lights so it would be very nice to have them recoglized in the light group