arendst/Tasmota #1785: Sonoff getting hot after running for only an hour

Shtutnik, 2018-02-04:
Hi,
I searched if someone else had run into this issue before but no luck.
I have two Sonoffs - one with the original Itead firmware and one where I put the latest firmware (5.11.1).
I touched them both and while the first Sonoff with the original firmware feels room temperature, the second one with the Tasmota firmware is pretty hot. I mean not melting hot, but quite warm. 
The warmth is coming from the bottom part, I think from where the ESP is located. 
Could it be that the firmware is making exhausting the ESP? Could it be because I haven't configured an MQTT broker yet? 
I also noticed that sometimes the physical button doesn't respond when I press it. it is as if the ESP is too busy.

Any help / guidance would very much appreciated.

reloxx13, 2018-02-04:
Did you set sleep?

I recommend sleep 150, but try it urself.

Shtutnik, 2018-02-05:
Hi,
Well yes that actually worked.. Now it seems as if it is in about the same temperature as the other Sonoff with the original firmware.

Thank you!

closed 2018-02-05 by Shtutnik

MetaImi, 2018-06-05:
where to set the sleep? :) I'm having the same issue

localhost61, 2018-06-05:
You didn't search a lot. Type "sleep" in the search field on top of this page and you will get 9 answers in the Wiki topic. The first one will answer to all your questions: [EnergySaving](https://github.com/arendst/Sonoff-Tasmota/wiki/Energy-Saving)

zuhair68, 2019-10-05:
Can I set sleep from Tasmota GUI?

localhost61, 2019-10-06:
Did you read the link of my previous post BEFORE asking?

zuhair68, 2019-10-06:
Thank you for your kind reply! People in every thread will have different knowledge in a specific subject, I asked because I didn't know where to input the command in GUI. You will be in the same position if you join us in an Autocad for architectural design forum.
If your reply don't add value, don't bother to reply, because you9not adding value anyway.

ascillato2, 2019-10-06:
Hi, this issue is very old. Now Tasmota has an automatic adaptive sleep time for energy saving that doesn't reduce the performance. There is no need to set a fixed value of sleep. There is no overheating due to software anymore. If you have any overheating, you should check the power regulator.

Thanks.