arendst/Tasmota #1787: After montes of using first block of flash erased with eeprom data

ChukXP, 2018-02-04:
@arendst 
First 0xFFF bytes from address 0x00000000 contains eeprom data with OTA link, SSID, passwords etc. Ofcourse device stop working.Version is 5.10.0a
Please check it with new versions.

arendst, 2018-02-04:
????????

ChukXP, 2018-02-04:
I flashed Sonoff basic with version 5.10.0a in 2017 december. It works fine until now. Few days ago it stops working (no any led lighting, no reaction on button pressing). I read flash backup and found that first 4096 bytes contains eeprom data with config data (OTA link, SSID, Password, mqtt settings etc.)

arendst, 2018-02-04:
Now that makes more sense to me. 

How did you flash your sonoff?
What flash mode did you use (DOUT)?
What flash size did you use (1M (no SPIFFS))?

ChukXP, 2018-02-04:
I flashed it via arduino software, mode DOUT, 1M no SPIFFS, all as described in wiki

arendst, 2018-02-04:
Pls provide output of command ``status 0`` once you have it running again.

ChukXP, 2018-02-04:
Here is after reviving, just flash, sensors not connected, MQTT not configured yet :
18:50:20 RSL: stat/sonoff/STATUS = {"Status":{"Module":1,"FriendlyName":"Sonoff","Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
18:50:20 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://my url/sonoff.ino.bin","Uptime":0,"Sleep":0,"BootCount":3,"SaveCount":6,"SaveAddress":"F6000"}}
18:50:20 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"5.10.0a","BuildDateTime":"2018-02-04T18:40:34","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
18:50:20 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId1":"MySSID","SSId2":"MySSID2T","TelePeriod":120,"SetOption":"00000009"}}
18:50:20 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":478,"Free":524,"Heap":25,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
18:50:20 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-3809","IPAddress":"192.168.254.43","Gateway":"192.168.254.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.254.254","Mac":"Here MAC","Webserver":2,"WifiConfig":3}}
18:50:20 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"domus1","MqttPort":1883,"MqttClientMask":"DVES_X","MqttClient":"DVES_2EAEE1","MqttUser":"DVES_USER","MAX_PACKET_SIZE":512,"KEEPALIVE":15}}
18:50:20 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Sun Feb 04 15:50:20 2018","Local":"Sun Feb 04 18:50:20 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":3}}
18:50:20 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2018-02-04T18:50:20"}}
18:50:20 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2018-02-04T18:50:20","Uptime":0,"Vcc":3.171,"POWER":"OFF","Wifi":{"AP":1,"SSId":"MySSID","RSSI":90,"APMac":"Here MAC"}}}

arendst, 2018-02-04:
Looks fine. You might want to load the latest development version 5.11.1i.

Also keep an eye on Status 1, SaveAddress as that should stay in the Fx000 region to not overwrite you block 0

closed 2018-06-19 by stale[bot]