arendst/Tasmota #1788: IR receiver

juanimoni, 2018-02-04:
I have attached a TSOP receiver to an ESP8266 and looking at what is sent by mqqt I see :

topic: "tele/ledStrip01/RESULT"
payload: "{"IrReceived":{"PROTOCOL":"SAMSUNG","BITS":32,"DATA":"7C"}}"
qos: 0
retain: false
_msgid: "957a0d53.a8eaf"

whatever button I press on the Samsung remote control, I get the same message. 
question : where is the actual code of the button ? 
thank you in advance for any help. I´m new to tasmota...

curzon01, 2018-02-04:
Can confirm that issue exists at minimum since 5.10.0 - but not in the current version 5.11.1i:

5.10.0 or 5.11.1: (wrong)
```
17:01:41 IRR: RawLen 68, Bits 32, Value 00000000, Decode 1587673860
17:01:41 MQT: dev/tele/tasmota/phesp03/RESULT = {"IrReceived":{"PROTOCOL":"NEC","BITS":32,"DATA":"7C"}}
```
5.11.1i: (OK)
```
17:03:53 IRR: RawLen 68, Bits 32, Value 00000000, Decode 1587673860
17:03:53 MQT: dev/tele/tasmota/phesp03/RESULT = {"IrReceived":{"PROTOCOL":"NEC","BITS":32,"DATA":"5EA1FB04"}}
```

juanimoni, 2018-02-04:
Definitely time to update !! 
was still on version 5.10.0. 
Thank you !!

curzon01, 2018-02-05:
Please close issue if solved

closed 2018-02-05 by juanimoni