arendst/Tasmota #1789: BIG issue Packet Loss when Emulation is enabled

savage872, 2018-02-04:
I have a couple of Sonoff basic and I was getting packet loss when I was pinging device of around 10-15%
Now I bought a Sonoff 4CH Pro, flashed it and kept the ping going and one the emulation was enabled I started having packet loss.
I hope Theo can have a look and sort it out...

arendst, 2018-02-04:
Short answer: Don't use emulation if you fear packet loss.

Emulation uses UDP uPnP and has to respond to every device sending it a M-SEARCH message. I discussed this before in other issues but I fear your network is already swamped with uPnP sources desperately searching for devices to monitor and/or control. To witness just enable more logging on your device and enjoy.

I see Sonos, Kodi, Routers fighting to get information from Tasmota. The poor thing can only service a few hungry uPnP devices and will likely loose some packets....

savage872, 2018-02-04:
Thanks for your quick reply! I appreciate it!

It's not that I fear packet loss, it makes the sonoff unresponsive. When you toggle switch or onboard button it doesn't respond.
I'm trying to use the sonoff's with Alexa and I don't really another way of doing it. If there is an alternative please let me know. (I can't use Home assistant cloud as it's not available in the UK)

How can you reduce UDP uPnP traffic on your network?

PS: Wornderful job with the Tasmota!

dYna77, 2018-02-07:
@savage872: you can use ioBroker and the built in MQTT.
It is a pretty goog software for automation anyway.

savage872, 2018-02-07:
I will give ioBroker a try. Thanks!

Can this be used with Alexa?

savage872, 2018-02-11:
I've sorted out my problem finally. I'm running on a pi3 hassbian (home assistant) with hue emulation enabled and it detect every component as a philips hue, so I've disabled emulation on the tasmota's and all is good! I hope this helps!

closed 2018-02-11 by savage872