arendst/Tasmota #1790: WS2812 Add option to color a range of pixels

martin072, 2018-02-04:
Would it be possible to add an option to set a color to a range of pixels on a strip?
The command  'LEDx #xxxxxx ' will do it for only one pixel, but I would like to control a range of pixels on the strip like this 
(preferably with the nice color fading option as well, look really nice on the whole strip when changing colours). 

Thanks!
Martin

arendst, 2018-02-04:
Command pixels lowers the amount of leds under control.

martin072, 2018-02-05:
yes, but that will set 0..<num pixels> I would be looking for an option to set pixels 0-30 in one color, 31-60 in another and 61-90 off for example.

arendst, 2018-02-06:
Too much hassle.

arendst, 2018-03-31:
And then it is implemented in the latest development version. See also #2303