arendst/Tasmota #1792: sonoff T1 uk - no wifi

cgorms, 2018-02-04:
Last night connected to a Sonoff T1 uk flash ota upgrade Tasmota connected to wifi network worked for 10 minutes set static ip via DHCP rebooted router now sonoff will not reconnect to wifi network.

Any advice would be great - how do i get wifi manager to work or put it into AP mode?

reloxx13, 2018-02-04:
https://github.com/arendst/Sonoff-Tasmota/wiki/Button-usage

cgorms, 2018-02-04:
@reloxx13 I'm struggle with the length of a short press should the light say come on twice and go off twice - sounds random but I've tried this several times

I take it I want 4 short presses?

cgorms, 2018-02-04:
Thanks for your help, I started WPS on my router sonoff T1 automatically connected to the wifi and I was able to managed etc from there - very happy now!!

closed 2018-02-04 by cgorms