arendst/Tasmota #1793: Device Crash after alot cmnd Requests

reloxx13, 2018-02-04:
Heya,

since 5.11.1i (maybe before, but not with stable 5.11.1) the device is crashing over ~115 request with 
http://SonoffIP/cm?cmnd=Status%200

It does not matter how fast you do those requests, its always crashing after that amount of requests.

i reproduce this with just spaming ctrl+r.

i noticed this by sonweb with a refrshtime of 15sec its crashing the devices after 1-2h.

Restart Reason says only "Software Watchdog".

i tried to reduce sleep from 150ms to 50ms but no change.

curzon01, 2018-02-05:
Try the same with same version you faulty testet but compile it using lib 2.3.0.

reloxx13, 2018-02-05:
5.11.1e working fine with 2.40 (20h runtime now).

arendst, 2018-02-06:
Pls try 5.11.1i with 2.3.0 too

reloxx13, 2018-02-07:
7h uptime, confirm fixed 👍

closed 2018-02-07 by reloxx13