arendst/Tasmota #1795: Sonoff ch4 does not react after flashing

kbaras, 2018-02-04:
Hello,

I have the following problem: I tried to flash a sonoff ch4 pro device (with stock firmware) with SONOTA to the latest tasmota firmware, but after it was done, it did not function properly: the first two/three leds are red and the buttons are not react as they should. For a few clicks on the e-fw button the first led goes on and off (with the blue led sometimes flashing, sometimes not), but after a few clicks it stops. I think fw flash was not okay...
Then I tried to erase everything with esptool, and also with a custom blank_bin file that I found to erase the whole data, and then re-flash tasmota, but the result is the same.

How can I be sure if erasing the data was successful, and then what goes wrong after flashing? 

I forgot to mention that flashing and erasing seems to go ok, without any errormessages.

Frogmore42, 2018-02-05:
It sounds like you have figured out how to connect to the serial pins on the Sonoff.  You now need to get a terminal program to see what the Sonoff thinks is going on and if anything is running there.  Tasmota prints a fair amount of information on the serial output.  So, it is running you should see some messages, which will tell you when things are wrong.  If you see no messages, you either don't have serial hooked up correctly or there is no functioning firmware in the Sonoff.

I use Tera Term, but other people like putty.  PlatformIO also has a serial terminal window that works well for this.

kbaras, 2018-02-05:
Yes, I tried serial monitor after erased the flash, but did not get the same output as in this article:
http://www.pratikpanda.com/completely-format-erase-esp8266-flash-memory/

Actually, my output is (when I power-on the device):
r␘␂␀l��r�␀�#␂�n��␌␘�␌␜��␜p�<���␀�8␂��ǒ��l␜p␌␘␌�nn�␂�;�nĒ��␌␛�␌b�$

And constantly the first three leds are red.

Frogmore42, 2018-02-05:
That output indicates your device is trying to connect to Theo's network.  Since you are not near there, it fails.

Have you made any changes to the code before flashing it?

I always configure at least my Wi-Fi network settings (in the code and recompile it) before flashing.  That makes it likely the device will connect and I can access it easily.

kbaras, 2018-02-05:
Hi!

Yes, I have tried it - not now, but earlier. But did not work as expected either (did not respond to push).
I will try once again with console output to see whats happening. 

What are these restarts anyway? It restarts every lets say half a minute

kbaras, 2018-02-05:
Ok, I think I know what the problem is: re-flashed with wifi-configured tasmota. I was able to open webpage, but it says: SONOFF basic module. 
How can tasmota distinguish between sonoff products? Am I missing some configuration?

kbaras, 2018-02-05:
I got the answer! It is configurable via the webpage. I chose the right module and vioila: everything is working!
Thank you for the info!

closed 2018-02-06 by arendst