arendst/Tasmota #1797: Multiple temperature sensors mqtt errors

isvedic, 2018-02-05:
Hi,

Noticed on my TH10 with 2x AM23xx sensors values sometimes mirrors each other:
10:03:22 MQT: tele/Vent-radna/SENSOR = {"Time":"2018-02-05T10:03:22","AM2301-04":{"Temperature":1.7,"Humidity":90.1},"AM2301-14":{"Temperature":1.7,"Humidity":90.1},"BME280":{"Temperature":26.0,"Humidity":43.4,"Pressure":1012.0},"TempUnit":"C"}
10:08:29 MQT: tele/Vent-radna/SENSOR = {"Time":"2018-02-05T10:08:29","AM2301-04":{"Temperature":1.8,"Humidity":90.1},"AM2301-14":{"Temperature":1.8,"Humidity":90.1},"BME280":{"Temperature":25.8,"Humidity":43.7,"Pressure":1012.0},"TempUnit":"C"}
10:13:35 MQT: tele/Vent-radna/SENSOR = {"Time":"2018-02-05T10:13:35","AM2301-04":{"Temperature":12.2,"Humidity":64.5},"AM2301-14":{"Temperature":1.7,"Humidity":90.1},"BME280":{"Temperature":25.3,"Humidity":44.8,"Pressure":1012.0},"TempUnit":"C"}
10:18:39 MQT: tele/Vent-radna/SENSOR = {"Time":"2018-02-05T10:18:39","AM2301-04":{"Temperature":12.1,"Humidity":64.5},"AM2301-14":{"Temperature":1.7,"Humidity":90.1},"BME280":{"Temperature":24.7,"Humidity":46.2,"Pressure":1012.0},"TempUnit":"C"}

As you can see on first 2 logs AM2301-04 had value of different sensor and after that it normalized.

On graph I can see it bounces like that for about hour and then true for couple more hours then starts again. No reboots in between. All occurrences seems random. Only  AM2301-04 takes values of AM2301-14, didn't noticed other way around. Sensors are on different GPIO

arendst, 2018-02-05:
Try to change lines 215 - 222 in file xsns_06_dht.ino from
```
  char temperature[10];
  char humidity[10];
  float t;
  float h;

  byte dsxflg = 0;
  for (byte i = 0; i < dht_sensors; i++) {
    if (DhtReadTempHum(i, t, h)) {     // Read temperature
```
to
```
  char temperature[10];
  char humidity[10];

  byte dsxflg = 0;
  for (byte i = 0; i < dht_sensors; i++) {
    float t = 0;
    float h = 0;
    if (DhtReadTempHum(i, t, h)) {     // Read temperature
```

isvedic, 2018-02-05:
Just changed and upgraded but same issue:
19:14:53 MQT: tele/Vent-radna/SENSOR = {"Time":"2018-02-05T19:14:53","AM2301-04":{"Temperature":1.1,"Humidity":90.1},"AM2301-14":{"Temperature":1.1,"Humidity":90.1},"BME280":{"Temperature":24.9,"Humidity":45.2,"Pressure":1009.9},"TempUnit":"C"}
19:19:53 MQT: tele/Vent-radna/SENSOR = {"Time":"2018-02-05T19:19:53","AM2301-04":{"Temperature":11.8,"Humidity":65.4},"AM2301-14":{"Temperature":1.1,"Humidity":90.1},"BME280":{"Temperature":23.7,"Humidity":47.6,"Pressure":1009.8},"TempUnit":"C"}
Interesting that UI web data is always looks correct and don't flap

Will leave it like that now, data can be accessed on: https://thingspeak.com/channels/419407

isvedic, 2018-02-09:
Looks stable last 24h with NAN under loop

isvedic, 2018-02-12:
been stable last 3 days but after reboot issue started again, both MQTT and domoticz:

13:01:20 MQT: tele/Vent-radna/SENSOR = {"Time":"2018-02-12T13:01:20","AM2301-04":{"Temperature":1.4,"Humidity":90.1},"AM2301-14":{"Temperature":1.4,"Humidity":90.1},"BME280":{"Temperature":19.4,"Humidity":57.6,"Pressure":999.4},"TempUnit":"C"}

13:06:23 MQT: domoticz/in = {"idx":25,"nvalue":0,"svalue":"12.0;64.5;1"}

13:06:23 MQT: tele/Vent-radna/SENSOR = {"Time":"2018-02-12T13:06:23","AM2301-04":{"Temperature":12.0,"Humidity":64.5},"AM2301-14":{"Temperature":1.3,"Humidity":90.1},"BME280":{"Temperature":19.3,"Humidity":58.0,"Pressure":999.5},"TempUnit":"C"}

13:11:25 MQT: domoticz/in = {"idx":25,"nvalue":0,"svalue":"1.3;90.1;3"}

13:11:25 MQT: tele/Vent-radna/SENSOR = {"Time":"2018-02-12T13:11:25","AM2301-04":{"Temperature":1.3,"Humidity":90.1},"AM2301-14":{"Temperature":1.3,"Humidity":90.1},"BME280":{"Temperature":19.3,"Humidity":58.2,"Pressure":999.3},"TempUnit":"C"}

13:16:28 MQT: domoticz/in = {"idx":25,"nvalue":0,"svalue":"1.4;90.1;3"}

13:16:28 MQT: tele/Vent-radna/SENSOR = {"Time":"2018-02-12T13:16:28","AM2301-04":{"Temperature":1.4,"Humidity":90.1},"AM2301-14":{"Temperature":1.4,"Humidity":90.1},"BME280":{"Temperature":19.2,"Humidity":58.6,"Pressure":999.3},"TempUnit":"C"}

isvedic, 2018-02-25:
last 7 days problem do not occurs when not accessing sonoff web interface

closed 2018-02-25 by isvedic