arendst/Tasmota #1798: Sonoff Dual R2 Button 1

cyberfly79, 2018-02-05:
Hello,

I have a Sonoff Dual R2. I would like to add 2 manual push buttons.
I set in config GPIO0 - Button 2. This works perfect. GPIO9 - ?? not already Button 1.
If I selected Switch 1, it works as long as I press it.
How do I changing the config, GPIO9 works like GPIO0

Thnks

calcipher49, 2018-02-05:
Pin button1 is GPIO0 and pin button2 is GPIO9
But in Tasmota software, button1 is assigned to intenal Button (GPIO10).
You can use Switch1 for GPIO0, but if you need Button1, you must change assignment in sonoff_template.h
- Find "Sonoff Dual R2" section, then line : GPIO_KEY1,        // GPIO10 Button on casing
- Replace GPIO_KEY1 By 0 (disable GPIO10) ou GPIO_USER (can be changed using web interface)
- Recompile and upload
Caution: internal button becomes unusable

schrej-zz, 2018-03-14:
Hi, configuration interface does not reflect, what is stated in the comment from my point of view:
"BUTTON 0 (= GPIO0) connected to GND controls relay 1 and BUTTON 1 (= GPIO9) connected to GND controls relay 2.
So the GPIO should be marked as "Button 0" (currently "Button 1") and GPIO 9 as "Button 1" (currently empty)