arendst/Tasmota #1799: [S20 / Request] More Control over Button-LEDs

TreuZuKaarst, 2018-02-05:
Use Case/Scenario:
I have a S20 with humidity-sensor controlling an electrical fan in my bathroom to control the humidity in the room. OpenHAB monitors this and controls all with a rule. So far - so good. Hence the S20 can be operated by a button you'll never know if the fan is running if it was triggerd and controlled by the rule or manually. As a result you'll never know if you have to turn it off manually or not.

Request:
I'll like to have more control over the green Button-LED. In this device always both LEDs (green=power / blue=relay) are on or off (in std mode). I'd like to have the opportunity to let the LED light permanently (=LEDState: x) if it was switched on manually (by button or OpenHAB GUI) an let it blink (=LEDState: blink) if it is in "automatic Mode" (controlling rule is active). So these states need to be switched by my OpenHAB rule via mqtt.

Is this already possible and I only didn't found the solution?

curzon01, 2018-02-05:
I have no idea to cover your blink request but why not using Blue LED, Blue+Green LED:

Set `LedState 0` to indicate blue LED only the Power on/off state (for manual on/off)
Set `LedState 1` to indicate blue and green LED the on/off state (for automatic on/off)

- If LED is off, switch is off
- If blue LED is on it was switched on manually
- If blue and green LED is on it was switched on automatic

TreuZuKaarst, 2018-02-05:
This will of course be a possible solution. But I wonder how to set LEDState 0 as default and persistent in this S20.

KPK75, 2018-02-05:
Or change LED1 to RELAY2 and control it by sending commands: POWER2 ON or POWER2 BLINK

curzon01, 2018-02-05:
@KPK75: Assigning PINs are limitied when you use Modul "Sonoff S20"

@TreuZuKaarst: You wrote 

> OpenHAB monitors this and controls all with a rule

You have to send proper "LedState x" command from your OpenHAB logic. If humidity is above limit set LedState to 1, otherwise to 0. The commands are persistent at all.
Use [MQTT](https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#mqtt) or [HTTP](https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#web)

TreuZuKaarst, 2018-02-05:
@curzon01
Thanks a Lot! you gave the right hints. Implemented it and it works like a Charm. 

Solved!

closed 2018-02-05 by TreuZuKaarst