arendst/Tasmota #1800: Issues with 433 remote

Shtutnik, 2018-02-05:
Hi everyone,
Love this firmware and already have big plans! Thanks alot for this awesome project!!

I have a Sonoff RF with the latest (5.11.1) firmware and am trying to figure out the responsiveness of the 433 remote.
I programmed the remote (pressed twice, got a red flash, pressed on the remote, got some more red flashes and the lights went on) and it works. 
Problem is - I can't program more than one button, or remote. With the original firmware I am able to do this. It is as if it only works with the last button learned. Although, when I press on the first button learned, the red led flashes, but it does not toggle the output power. 
Also, it responds in a different manner than the original firmware. With the original firmware, I only need to press the button on the remote for a short period to get it to work, but with the Tasmota I need to hold it for about a second or two and it doesn't always work.

Can anyone point me to the relevant code in the firmware where I can try to figure out why this is happening? Where are the learnt codes stored? Is it in the 433 module itself? If so, why doesn't it trigger the esp on ALL codes learnt? (It just flashes the red led). Where is the handling of the 433 trigger in the code? 

Thanks again!!

arendst, 2018-02-05:
Tasmota does nothing regarding 433 reception. This is done by a separate microcontroller not under control of the ESP8266 hardware. All this microcontroller does is lowering the button input on the ESP8266.

If you are unable to learn more than one remote then that's dealt with by this controller. You won't find any specific Sonoff RF 433 code in tasmota.

Shtutnik, 2018-02-05:
Oh thank you!
I had a feeling that this is handled in a sepperate controller. 
I just want to be more clear - I think that the 433 controller is able to learn more than one code, since I do see the red led flashing when I press either learnt codes. The problem lies in that only the last code is able to trigger the esp. Is it possible that the code that senses the button is very strict? And that only the last learnt code lowers the button long enough? Could you perhaps point me to the code that handles this? I mean detection whether the button is pressed.
Many thanks!!

Shtutnik, 2018-02-05:
Ok, I found the relevant code at around line 1940. So turns out that `SetOption13 on` helps make the sonoff respond immediately and not count the presses (which seems to take 0.5 a second longer). Which is better for my use case. 
But the other problem still exists.. :( Only the last learnt code is responding, all other learnt codes seem to turn the red led on, but don't trigger the button as "pressed". 
Can someone measure what is happening there? maybe the check of `< 99` isn't enough?

Also, could it be that this code (at line 2555):
```
for (byte i = 0; i < MAX_KEYS; i++) {
    if (pin[GPIO_KEY1 +i] < 99) {
      pinMode(pin[GPIO_KEY1 +i], (16 == pin[GPIO_KEY1 +i]) ? INPUT_PULLDOWN_16 : INPUT_PULLUP);
    }
  }
```

needs to use `devices_present` instead of `MAX_KEYS`?

MatzeJoerling, 2018-02-05:
i have the bridge, and it works with more than one remote. 
perhaps i am totally wrong, but 
did you learn the codes with "rfkeyX 2", where X is number from 1 to 16 for your varius remote keys?
If you have 4 Channel remote, 8 Buttons you have to learn "rfkey1 2" to "rfkey8 2" for all of them.

Shtutnik, 2018-02-05:
@MatzeJoerling 
I think that those commands are specific to the bridge. 
The Sonoff RF handles RF commands internally (within the 433 module) and simulates a button press when a learnt code is detected. I think that the problem may be with the EFM8BB10F2G chip (which translates the 433 module output to a button press). I think that for previously learnt codes it simulates the button pressed in a different manner than the last learnt code. I have no proof currently, but will do some measuring later this week. 
If anyone has some experience with this, or has a Sonoff RF unit and a remote with more than 1 button, could you please verify if this replicate son your end?
Thanks so much

MatzeJoerling, 2018-02-06:
i was totally wrong, sorry for that. I have the RF Bridge, not the "Sonoff RF".