arendst/Tasmota #1803: Question about MQTT subscriptions

SGF-Lon, 2018-02-05:
Looking at some tracing logs, I can see the following MQTT subscriptions after a restart.  Anyone know what the purpose of MQTT topic #012 (see last log entry)  is ?  I'm not even sure if it's formatted correctly (i.e. is it missing a '/' ) ??

Anyone have any insight? 

The device is a Sonoff Touch 2 channel & the firmware is 5.10.0

Feb  5 14:10:14 Dev_7588 ESP-MQT: Connected
Feb  5 14:10:14 Dev_7588 ESP-MQT: Sonoff/tele/Sonoff_7588/LWT = Online (retained)
Feb  5 14:10:14 Dev_7588 ESP-MQT: Sonoff/cmnd/Sonoff_7588/POWER =
Feb  5 14:10:14 Dev_7588 ESP-MQT: Subscribe to Sonoff/cmnd/Sonoff_7588/#
Feb  5 14:10:14 Dev_7588 ESP-MQT: Subscribe to Sonoff/cmnd/sonoffs/#
Feb  5 14:10:14 Dev_7588 ESP-MQT: Subscribe to cmnd/DEV_7588#012/#

ionciubotaru, 2018-02-07:
Tasomta sends an empty message to `Sonoff/cmnd/Sonoff_7588/POWER` to be sure that any prevoius retain `power state` message is deleted. Sending empty messages is the only way you tell the MQTT server to delete a retained message.

arendst, 2018-02-07:
Actually the message POWER= without result is a request of the HomeBridge community to let HomeBridge know the basic command to toggle power for this device.

The last subscription is the fallback topic made up from the default ClientId. Where the #12 comes in beats me but it should probably be hex 0C.

A "normal" log 3 looks like this:
```
10:27:15 MQT: Connected
10:27:15 MQT: tele/pzem/LWT = Online (retained)
10:27:15 MQT: cmnd/pzem/POWER = 
10:27:15 MQT: Subscribe to cmnd/pzem/#
10:27:15 MQT: Subscribe to cmnd/sonoffs/#
10:27:15 MQT: Subscribe to cmnd/DVES_0CA8BA/#
```

SGF-Lon, 2018-04-24:
Thanks for the information @arendst @ionciubotaru . The POWER command isn't a problem - although it does cause Openhab to hiccup.  However, if the last subscription is the fallback topic - then it clearly doesn't use the MQTT_FULLTOPIC from user_config.h.   This should probably be highlighted in the wiki & documentation.  The #012 is still a mystery and worth investigating as it makes it difficult to use the fallback topic as documented with a larger number of devices.

closed 2018-06-23 by stale[bot]