arendst/Tasmota #1805: What is the amount of WiFi transmissions during standby ?

fabio-cecamore, 2018-02-05:
Hello,
I would like to know if during the standby period of the device if it produces WiFi transmission (and if so, how much) considering that if there should be more than 50 devices installed in a house, the transmission could end up being unsafe for human exposure.

Cheers

zybeon, 2018-02-05:
Short answer is no. WiFi does not pose any risks due to it's low power output, especially by the ESP8266 units.

The very low output power and frequencies of WiFi are no where near enough to be damaging. There are studies out their that debate this but more research would be required than what is already out there to determine if WiFi transmission are dangerous in any way. You also have to take in the account the output power and distance from the emitter from all those studies. Coming from the military and being around incredibly high power unidirectional transmitters, safe distance is considered 10 feet from something outputting in the SHF range and at 100 watts of power. The frequency of WiFi 2.4/5.8GHz is in the high UHF range(300MHz-3GHz) and very low SHF range(3GHz-30GHz). Normal output power is well below 1W and normally less than 100mW. Since the vast majority of WiFi devices are omnidirectional, radiation drops very dramatically with distance. Since it is easier to test to most devices 

> "The radio waves produced by Wi-Fi and smart meters are very low power, much lower than those given off by mobile phones, and well within international guidelines. The evidence to date suggests exposures to the radio waves produced by smart meters and Wi-Fi do not pose a health risk."
[Cancer Research UK (Oct 2016)](http://www.cancerresearchuk.org/about-cancer/causes-of-cancer/cancer-controversies/mobile-phones-wifi-and-power-lines)

> "WiFi operates in the 2 to 5 GHz range -- part of the microwave portion of the electromagnetic spectrum. This is in the same part of the spectrum where cell phones operate so I may refer to WiFi or cellphone electromagnetic radiation interchangeably. These are radio waves -- no different from those used to broadcast television programs, except that they are higher in frequency. They aren't nearly as high a frequency as visible light, and no one worries about getting cancer from visible light (ultraviolet light, on the other hand, causes skin cancer, but this is the minimum energy necessary to cause ionizations that can cause breaks in strands of DNA, which is the mechanism by which cancer cells can be created). There is no credible evidence that non-ionizing radiation has any adverse health effects at all. There is no radiobiologic mechanism that could explain such an association -- and absolutely no scientifically valid evidence that this has ever happened."
[Dr, Gary Larson, Medical Director at Procure Proton Therapy Center (May 2016)](https://www.forbes.com/sites/quora/2016/05/19/a-radiation-oncologist-says-everything-you-need-to-hear-about-wifi-and-cancer-risk/#495038947267)

> "To date, the only health effect from RF fields identified in scientific reviews has been related to an increase in body temperature (> 1 °C) from exposure at very high field intensity found only in certain industrial facilities, such as RF heaters. The levels of RF exposure from base stations and wireless networks are so low that the temperature increases are insignificant and do not affect human health.
>
>The strength of RF fields is greatest at its source, and diminishes quickly with distance. Access near base station antennas is restricted where RF signals may exceed international exposure limits. Recent surveys have indicated that RF exposures from base stations and wireless technologies in publicly accessible areas (including schools and hospitals) are normally thousands of times below international standards."
[World Health Organization (May 2006)](http://www.who.int/peh-emf/publications/facts/fs304/en/)

Also check: [IEEE (2006) IEEE C95.1-2005 "IEEE Standard for Safety Levels with Respect to Human Exposure to Radio Frequency Electromagnetic Fields, 3 kHz to 300 GHz"](http://ieeexplore.ieee.org/document/1626482/)

Looking at the FCC testing of the [ESP8266](https://fccid.io/2AHMR-ESP12S/RF-Exposure-Info/SZEM160300123402-3087392), it outputs 14.289mW to the antenna. More testing can be found here: https://fccid.io/2AHMR-ESP12S/Test-Report/SZEM160300123401-rev-3087394

To conclude, I would say cellphones are way more "dangerous" than WiFi devices due to the higher potential power output of 250mW as a general estimate of their highest power output setting and I would never hesitate to hold one next to my head all day. They fluctuate their power output in response to the tower sending it back it's signal level to save battery. Newer modulation techniques are lowering the power output required each iteration. Which means over time technology will lower their a

fabio-cecamore, 2018-02-07:
Thank you for your answer, unfortunately you have not addressed my specific question which is: what is the measurement of frequency transmitted by the device in standby mode (due to WiFi and MQTT protocols)? Thank you in advance for your reply.

Frogmore42, 2018-02-08:
Are you asking the frequency of the RF that are used?  If so that is in the 2.4GHz ISM radio band.
Are you asking the number of times and for how long does the Wi-Fi radio come on as the device is operating normally?
Are you asking the power of the RF transmissions that occur?
Why do you want to know, ie what is your concern?

@zybeon answered what most would think was your real question.  

>  the transmission could end up being unsafe for human exposure.

For the other two questions you might have (that I listed above) I don't know and I doubt that many people do.  All of them could be measured on a actual device, but would require some specialized equipment and a fair amount of time to get an accurate answer.

fabio-cecamore, 2018-02-09:
Hi Frogmore42, thank you for your contribution.

> For the other two questions you might have (that I listed above) I don't know and I doubt that many people do. All of them could be measured on a actual device, but would require some specialized equipment and a fair amount of time to get an accurate answer.

I think it is not unlikely that someone will be able to answer my two questions. The MQTT protocol should be fully understood by the developer of this firmware, and maybe, also how the ESP8266 (or the generic layer 0 WiFi protocol) works.

I am willing to accept that my "50" future sonoffs transmit during use, but I would like to know if there are continuous transmissions even during times of inactivity, which would make them, in my opinion, less safe.

Frogmore42, 2018-02-09:
Define continuous.  Both Wi-Fi and MQTT require regular communications to be considered alive.  Neither is continuous.  You are asking things that have nothing to do with Tasmota, but do with the underlying projects upon which Tasmota builds.  There is a lot of information available on the details of MQTT and Wi-Fi on the internet.  

It is up to the user who configures Tasmota to determine how frequently Tasmota makes MQTT messages.  But, normal station to access point communications are determined by the Wi-Fi protocol specification.  It to is a low duty cycle, but does require transmission on a periodic basis.  Even more than a decade ago, when I was on the Wi-Fi Alliance and worked on Wi-Fi routers, I didn't have that information at the top of my head.  If you really want the information you are going to have to dig deeper.  

I won't argue with you on the safety of Tasmota.  Everyone has to make the decision for him/herself.  I have many Tasmota devices in my house and am not particularly concerned about it.  In fact, I am glad, since they provide value to me.  I don't believe you will find a better answer on this forum.

davidelang, 2018-02-09:
what do you mean by 'standby'

if you mean times when the tasmota firmware is not transmitting, then there are 
the standard wifi beaconing and keepalive handshaking (standard for anything 
that does wifi)

if you are excluding that, then you will need to dig into the FCC certification 
records for the ESP8266 module, it's not something anyone here has bothered to 
do.

If you are talking about transmissions controlled by the firmware, there are the 
commands, the command responses, and the telemetry broadcasts.

It's not clear what you are looking for.

fabio-cecamore, 2018-02-10:
Thanks to both of you for the comments.

What I'm looking for is trying to understand if during times of inactivity (when I do not press a switch for example) how many transmissions sonoff generates: among those MQTT and those that are part of the WiFi protocol.

> if you mean times when the tasmota firmware is not transmitting, then there are
the standard wifi beaconing and keepalive handshaking (standard for anything
that does wifi)

Yes, that is what I mean. If I'm not wrong the beaconings are from the Access point, not from the clients, but what about the keepalive handshaking ?

I am aware of being a bit offtopic but maybe it could be an interesting topic for other users too.
Cheers

Frogmore42, 2018-02-11:
Its not just off topic, there is unlikely to be anyone on this forum that knows how to answer your question(s)off the top of their head.  As I said earlier, even when I was in the business over a decade ago, I would not have been able to answer the question simply, since it really isn't a simple question and involves a LOT of details.  If you are really interested, do some searching on the internet for Wi-Fi protocol details and MQTT protocol.  I am sure if you look hard enough you can find more information about the frequency of transmissions, duration is going to vary depending on negotiated data rate, but I don't know what it will take to answer your real question.  The protocols allow for some choices so the real answer is, it depends.  In any case it is unlikely that they are active for even 10% of the time for "housekeeping" but it might be closer to 1%.  And, you are correct that the beacon is sent from the AP to all stations.

davidelang, 2018-02-11:
when no actions are taking place on the device, it sends a mqtt message out 
every teleperiod seconds

other than that, there is whatever beaconing and keeplive stuff that's common to 
any wifi device.

fabio-cecamore, 2018-02-16:
> other than that, there is whatever beaconing and keeplive stuff that's common to
any wifi device.
That's the problem, it is not easy to find at information about this.

Frogmore42, 2018-02-16:
It is By Design that "housekeeping" is a very small percentage of total bandwidth (and therefore time).  Since no one NEEDS to know more detailed information to use Wi-Fi, if you really, really, really, really want to know it, you will need to dig very deep to find the places where the people who design the radios and software for Wi-Fi hang out and ask them (I am pretty sure this is not that place).  Or, you can dig through the specifications and see if you can figure it out.  Here is a pointer: http://www.radio-electronics.com/info/wireless/wi-fi/ieee-802-11-standards-tutorial.php

davidelang, 2018-02-17:
On Fri, 16 Feb 2018, fabio-cecamore wrote:

>> other than that, there is whatever beaconing and keeplive stuff that's common to
>> any wifi device.

> That's the problem, it is not easy to find at information about this.

why do you care?

or more importantly, why should any of the rest of us care? It's not something 
we can change.

David Lang

ascillato, 2018-02-17:
Hi,

The _ESP8266_ is developed to be an IoT Device for low energy consumption, so, it has sleep capability that also keeps the wifi connection alive.

From the ESP8266 datasheet on https://www.espressif.com:
> **1) Modem-Sleep** requires the CPU to be working, as in PWM or I2S applications. According to 802.11 standards (like
> U-APSD), it saves power to shut down the Wi-Fi Modem circuit while maintaining a Wi-Fi connection **with no data transmission**. E.g. in DTIM3, to maintain a sleep 300ms-wake 3ms cycle to receive AP’s Beacon packages, the current is about 15mA.
>
> **2) During Light-Sleep**, the CPU may be suspended in applications like Wi-Fi switch. Without data transmission, the Wi-Fi Modem circuit can be turned off and CPU suspended to save power according to the 802.11 standard (U-APSD). E.g. in DTIM3, to maintain a sleep 300ms-wake 3ms cycle to receive AP’s Beacon packages, the current is about 0.9mA.
> 
> **3) Deep-Sleep** does not require Wi-Fi connection to be maintained. For application with long time lags between data transmission, e.g. a temperature sensor that checks the temperature every 100s ,sleep 300s and waking up to connect to the AP (taking about 0.3~1s), the overall average current is less than 1mA.

So, during sleep there is no transmission.

Besides, [this article](https://www.imgtec.com/blog/how-low-powered-wi-fi-sensors-are-the-future-of-iot/) explains that the _keep alive messages_ should be transmitted **every 30 seconds**

Also, you will have the transmission of the _Telemetry MQTT message_ as defined on `TELE_PERIOD` in _user_config.h_:
```
// -- MQTT - Telemetry ----------------------------
#define TELE_PERIOD            300               // [TelePeriod] Telemetry (0 = disable, 10 - 3600 seconds)
```

_**So, while using sleep, you are transmitting every 30 seconds on Stand By.**_

Also from the ESP8266 datasheet, the max transmiting power is 21dBm

And from [WiFi Transmit Power Calculations Made Simples](http://www.digitalairwireless.com/wireless-blog/t-eirp/wifi-transmit-power-calculations-made-simples.html) we can convert 21dBm to 128mW.

So, on Stand By, you should have as a maximum a peak of 128mW of EM every 30 secs. This peak will depends on the distance to router, position, etc. So, it should be less than that.

Now, about the health risks. You have to measure the Dose of Power you receive. The dose depends on the distance to the transmitter and the time you are exposed. The dose is inversely proportional to the distance, meaning that if you double the distance the dose is 1/4 of the previous dose. The time is directly proportional.

So, lets say that mqtt is also 3ms TX as the keepalive TX.

TxTime/hour = 120 keepaliveMSG/hour + 12 mqttMSG/hour
TxTime/hour = 132 TxMSG of 3ms = 0.00011 TxTime/Hour

Power(hr) = 128mW*TxTime = 0.01408mW/hr

Now, we can assume that you are at 2 meters from the Sonoff, so Power/hr = 0.00352mW/hr

And also that you are receiving just part of the radiation (because the EM wave will go in several directions). Taking a simplified spherical wifi signal, the surface at 2m is A = 4πr2 = 50.27 m2

Using [body surface calculator](http://www.calculator.net/body-surface-area-calculator.html) and assuming a male of 80Kg and 1.7m height, the body's skin surface exposed is 1.92m2 = 19200cm2

So, you are receiving 1.92/50.27 x 100 = 3.81 % of the total signal passing thru you.

        Power(hr) = 0.00013mW/hr

         So, Power exposed to body 0.007 uW/cm2

Hence, using this [thesis](http://www.theseus.fi/bitstream/handle/10024/70736/karki_ramesh.pdf?sequence=1), that **dose of 0.007 uW/cm2 can just give headaches or may be nothing.**

If you want to scalate this for 50 devices, take into your analisys that all the devices will be in different parts of the house, so, more or less is going to be that you are just a 2m from just 1 device and the rest of the devices can be ignored because of their distance.

Please, remember that the body is receiving signals all the time and it is prepared for that. So, for small doses the body repair itself and no damage occur. You should be aware of high rate doses as sleeping with a cellphone close or living under a power transformer or high power line, or close to a cell phone tower.

Hope this very simplified analysis help a bit on this issue. All the information used, are on the links and open to debate.

davidelang, 2018-02-17:
add to all that the the power hitting you isn't all going to absorbed by you, a 
significant amount of it passes through you, reducing the impact even further.

David Lang

ascillato, 2018-06-02:
Hi @fabio-cecamore

If you think your question has been addresses, please, close this issue. Thanks :+1:

closed 2018-06-02 by fabio-cecamore