arendst/Tasmota #1806: Issue detecting serial por on mac

luisfpinto, 2018-02-05:
Hello,

I'm trying to connect the sonoff device to my MAC with a TTL-Usb connector. 

The sonoff device turns on properly if I don't flash it so I suppose pins are connected okay. 

I've tried to switch TX & RX but I'm always getting the same error:
```
Looking for upload port...
Error: Please specify `upload_port` for environment or use global `--upload-port` option.
```
Any help on this will be really appreciated

davidelang, 2018-02-05:
what version OS, there have been a lot of problems reported using high sierra, 
and work-arounds have been posted (search closed issues)

johngouk, 2018-02-06:
Yes, I had to modify my drivers on High Sierra. Do you know which USB-Serial chip you're using? i now use the standard Apple USB-Serial driver for FTDI, and I'm using the v1.4 for CH340 chips. Here's [my blog about the FTDI fix](http://arduinoarsingabout.blogspot.co.uk/2018/01/plan-b-flashing-sonoff-devices-using.html) and I found [the 1.4 CH340 drivers here](http://www.wch.cn/download/CH341SER_MAC_ZIP.html). If you're using the Silicon Labs CP2012/4, their drivers are on their website. Have fun.

luisfpinto, 2018-02-06:
Hey! I figure it out! I was missing installing the drivers for the USB serial. Once I installed them everything was working as a charm! Thank you so much for the support! I'm closing the issue now

closed 2018-02-06 by luisfpinto