arendst/Tasmota #1808: Webserver and emulation issues

M-Rod89, 2018-02-06:
Hey all I'm having issues with the web server failing and emulation sometimes not working. I've disabled as much I think I can and upgraded sitting at 24k memory free and then eventually I assume run out of memory because web server doesn't respond. Can I free up anymore? or am I missing something? All I need is just the webserver and emulation.

savage872, 2018-02-07:
The problem that I described had more or less the same thing. I found that I was loosing packets when I was pinging the sonoff and it turned out to be the emulation, if you disable the emulation then the web server should work just fine, Theo said it's because of to much uPnP traffic on your network. I'm still having this problem because I'm using emulation (10-15% packet loss). Hopefully there is a fix.

Hope it helps!

savage872, 2018-02-11:
I've sorted out my problem finally. I'm running on a pi3 hassbian (home assistant) with hue emulation enabled and it detect every component as a philips hue, so I've disabled emulation on the tasmota's and all is good! I hope this helps!