arendst/Tasmota #1810: No output on BLUE channel

lance36, 2018-02-06:
Hey, 

i connected in uart and flashed tasmota, setconfig 15 1, yadda yadda. i can control all the color but blue:

```
Color 1 or FF0000 is OFF
Color 2 or 00FF00 is RED
Color 3 OR 0000FF is GREEN

```

any idea on how to troubleshoot a pwm issue? should be coming from `GPIO12`

reloxx13, 2018-02-06:
device?

lance36, 2018-02-06:
It's the magic

lance36, 2018-02-06:
It's the MagicHome  LED strip Controller, the variant without the exposed pins as described at the and of:

https://github.com/arendst/Sonoff-Tasmota/wiki/MagicHome-LED-strip-controller
[image]

reloxx13, 2018-02-17:
https://github.com/arendst/Sonoff-Tasmota/wiki/MagicHome-LED-strip-controller

lance36, 2018-02-17:
Yeah, i had to change this part in the sonoff_template.h:

```
  { "MagicHome",       // Magic Home (aka Flux-light) (ESP8266) - https://www.aliexpress.com/item/Magic-Home-Mini-RGB-RGBW-Wifi-Controller-For-Led-Strip-Panel-light-Timing-Function-16million-colors/32686853650.html
     0,
	 0,
     GPIO_LED1_INV,    // GPIO02 Blue onboard LED
     0,
     GPIO_USER,        // GPIO04 IR receiver (optional)
     GPIO_PWM1,        // GPIO14 RGB LED Red
     0,
     0,
     0,
     0,
     0,
     0,                // Flash connection
     GPIO_PWM2,        // GPIO05 RGB LED Green
	 GPIO_PWM3,        // GPIO12 RGB LED Blue
     GPIO_USER,        // GPIO13 RGBW LED White (optional - set to PWM4 for Cold White or Warm White)
     0, 0, 0
  },

maybe i have a different revision of the board?
```

lance36, 2018-02-17:
[image]
[image]

lance36, 2018-02-17:
would love to be able to set RED\BLUE\GREEN pins on the GUI

lance36, 2018-02-17:
@reloxx13 i did read that.
in fact, I'm adding to it :)

stewface, 2018-10-06:
Can you change the pinout config via console? I just unsoldered everything :(

niklaskr, 2018-10-25:
Forget what i said.
You can use premade 37 Arilux LC01 device settings.
I found it in the sonoff_template.h becuase it uses also an esp8285
Mine works perfectly fine even with hue, rgb, brightness etc.

ozett, 2018-12-17:
sorry to bother in,
but i experience same behaviour. 
and the bug(") is not really fully solved..

as i have "a clone labelled “ZJ-ESP-IR-F V1.1”  of the MagicHome https://github.com/arendst/Sonoff-Tasmota/wiki/MagicHome-LED-strip-controller the configuration-hints in the wiki are hard to find:

[image]

as with this version of hardware and tasmota 6.4 i still have the same trouble: no BLUE lights...

it tried a lot of configs for the PMWs on different GPIOs with no success.

as for the other hardware-templates it is not always clear if they work on the esp 8285,
it carefully tried not to flash them. 

my problem is still not solved : https://github.com/arendst/Sonoff-Tasmota/issues/1867#issuecomment-447797679

but the thread has lots of same led-sorting solutions, maybe one could create another hardware-template?