arendst/Tasmota #1811: EnergyReset 3 don't work

aferraz, 2018-02-06:
Hi @arendst 
 "EnergyReset 3" don't give the expected results, it seems reset only Today

21:31:58 MQT: stat/sonoff2/STATUS10 = {"StatusSNS":{"Time":"2018-02-06T21:31:58","ENERGY":{"Total":183.360,"Yesterday":2.788,"**Today":1.841,**"Power":73,"Factor":0.73,"Voltage":225,"Current":0.449}}}
21:32:04 CMD: **EnergyReset 3**
21:32:04 MQT: stat/sonoff2/RESULT = {"EnergyReset":{"Total":181.522,"Yesterday":2.788,"**Today":0.003**}}

Thank you
 Alessandro

arendst, 2018-02-06:
Well actually that's what you told it to do ;-)

EnergyReset command syntax has changed. Although the wiki is updates I'll enhance it a little.

You should now use 
- ``EnergyReset1 <value>`` to change today (equal to ``EnergyReset <value>`` as you entered)
- ``EnergyReset2 <value>`` to change yesterday
- ``EnergyReset3 <value>`` to change total

aferraz, 2018-02-06:
Thanks @arendst

aferraz, 2018-02-06:
Hi @arendst,
 sotto but EnergyReset3 not always work.
I tried on different POWs, firmware version 5.11.1l:

22:54:14 MQT: stat/sonoff3/STATUS10 = {"StatusSNS":{"Time":"2018-02-06T22:54:14","ENERGY":{"**Total":175.29**,"Yesterday":3.05,"Today":1.69,"Power":0.00,"Factor":0.00,"Voltage":223.60,"Current":0.00}}}
22:54:22 CMD: **EnergyReset3 0**
22:54:22 RSL: Received Topic /EnergyReset3, Data Size 1, Data 0
22:54:22 RSL: Group 0, Index 3, Command ENERGYRESET, Data 0
22:54:22 MQT: stat/sonoff3/RESULT = {"EnergyReset":{"Total":1.69,"Yesterday":3.05,"**Today":1.69**}}
22:54:22 CFG: Saved to flash at FB, Count 454, Bytes 1648
22:54:28 CMD: status 10
22:54:28 RSL: Received Topic /status, Data Size 2, Data 10
22:54:28 RSL: Group 0, Index 1, Command STATUS, Data 10
22:54:28 MQT: stat/sonoff3/STATUS10 = {"StatusSNS":{"Time":"2018-02-06T22:54:28","ENERGY":{"**Total":175.29**,"Yesterday":3.05,"Today":1.69,"Power":0.00,"Factor":0.00,"Voltage":223.60,"Current":0.00}}}

Thanks 
 Alessandro

arendst, 2018-02-07:
Two items:
- the result of ``EnergyReset3 0`` at 22:54:22 becomes 1.69 as Total is the sum of Total energy used at 00:00 added with the today usage. So it becomes equal to today = 1.69
- why it is still 175.29 at 22:54:28 might be caused by changes I did within the 5.11.1i release around energy display. In my situation it works as expected:

```
10:14:44 CMD: status 10
10:14:44 MQT: stat/pzem/STATUS10 = {"StatusSNS":{"Time":"2018-02-07T10:14:44","ENERGY":{"Total":0.015,"Yesterday":0.002,"Today":0.003,"Power":0.0,"Factor":0.00,"Voltage":218.7,"Current":0.000}}}
10:15:44 CMD: energyreset3 0
10:15:44 MQT: stat/pzem/RESULT = {"EnergyReset":{"Total":0.003,"Yesterday":0.002,"Today":0.003}}
10:15:52 CMD: status 10
10:15:52 MQT: stat/pzem/STATUS10 = {"StatusSNS":{"Time":"2018-02-07T10:15:52","ENERGY":{"Total":0.003,"Yesterday":0.002,"Today":0.003,"Power":0.0,"Factor":0.00,"Voltage":218.8,"Current":0.000}}}
```

aferraz, 2018-02-07:
@arendst after midnight has been reset. Thanks

closed 2018-02-07 by aferraz

iz3man, 2018-04-26:
doesn't work here either. when i enter `energyreset3 0` to reset total kWh to zero, nothing happens. it just shows the counters. tasmota version is 5.5.1
```17:57:15 CMND: energyreset3 0
17:57:15 MQTT: stat/waermepumpe/RESULT = {"EnergyReset":{"Total":310.927, "Yesterday":0.019, "Today":0.014}}```
shall i wait until tomorrow to make the changes happen? is this how it's supposed to work?

iz3man, 2018-04-28:
waited a day, but nothing changed. any ideas what to do? summer season starts and i'd like to reset the pool's heat pump counter.

arendst, 2018-04-28:
@iz3man you're complaining about a problem in an old version (5.5.1) which has probabaly long been solved in later versions. 

Upgrade to 5.12.0 (or 5.12.0m) and try again.

iz3man, 2018-04-28:
@arendst you're absolutely right. my fault. i installed some new sonoffs and totally forget to upgrade the ones that show this issue (the only 2 POWs i have). i'll instantely upgrade. 
thanks!!