arendst/Tasmota #1812: help wanted - Tasmota version 5.11.1h running in aLx-wifi-03o Wireless WIFI Smart switch (3 buttoms

ctripodi, 2018-02-07:
Hi all,
After many attempts I could save the Tasmota firmware in a Lx-wifi-03o Wireless WIFI Smart switch (3 buttoms) [https://fccid.io/2ANKOLX-WIFI-03O/User-Manual/User-manual-3554904](url)
With the original firmware, When a switch is OFF, the touch buttons are blu. When the switch is ON, the touch button is turned to red. Heach touch buttom works in the same way with the original firmware.
Using tasmota, configured like an "Sonoff T1 3CH", with all the switches OFF, all the touch buttoms are blue. But when one of the touch switches is turned ON, the led of this touch buttom is turned to red, but the rest are powered off (just the led of the touch buttom, however they should still on and in blue)
The device works fine from the point of view of the rele circuits, the ligths connected to the switch works fine according to the switch status.
Had someone the same problem? How I can solve it?
I have selected the correct module?
More photos of this device here [https://fccid.io/2ANKOLX-WIFI-03O/Internal-Photos/Internal-photo-3554919](url)
Thanks
[image]
[image]

DaPeace, 2018-03-16:
@ctripodi how did you flash the device? got two of those today. Did you solder the uart directly to the esp or are there any pins i could use?

sabag, 2018-03-25:
i also would like to flash the device to use my own MQTT server.
does someone have some information on it (searched google for the last two days now) ?

frawau, 2018-03-31:
I have got a different rev of that board , I would still be interested in knowing how @ctripodi managed to flash the firmware... please

jimkernsjr, 2018-04-12:
Theres a JTAG right on the front of the board.  Its a ESP8285 module, just check the Itead web site for the pinout:
https://www.itead.cc/psa-01.html
It's an PSA-series module, you have TX/RX/VCC/Ground on the pinout, you just need to ground GPIO0 when you apply power and it's ready to take a program.
I don't understand what the problem is?

ctripodi, 2018-04-12:
Hi @frawau I flashed the firmware using this wiring connection
[image]

frawau, 2018-04-13:
My board is probably too different

[image]

sabag, 2018-04-13:
finding the pins to flash is just half of the problem. the rest of the problem is whether the buttons will work and lit as expected.
so did someone actually made it through and see if its working ?

frawau, 2018-04-13:
Indeed, but my guess is that they were very lazy and did not want to modify the Sonoff firmware. In any case I need to flash to test....

j2mc, 2018-04-15:
I've got the same board, I was able to flash it with sonota without issue.  Haven't been able to play with it much yet, but seems to be working properly so far.

sabag, 2018-04-15:
@j2mc , can you tell me what was the option you chose in tasmota and what pin did you grounded to enter to flash mode? 
thanks in advance

j2mc, 2018-04-15:
For SonOTA you don't need to enter flash mode, just follow their instructions(use the python version, the exe didn't work for me).  Once running tasmota I chose the T1 3 channel, and set LEDState 0 and SetOption13 1.  After that everything seems to work great.

sabag, 2018-04-20:
@j2mc  thank you so much, it worked smoothly. I only configured "module" as "sonoff t1 2ch", but i could not find the LEDState and SetOption13 in the web console. but thi most importat thing is that the buttons are working and their led reflect the correct state.

j2mc, 2018-04-20:
Easiest way to set those options is open the "Console" in the web interface and enter those commands there.  "SetOption13 1" will make it respond immediately when touched, "LEDState 0" will turn off the LEDs when off(red when on).  More info [here](https://github.com/arendst/Sonoff-Tasmota/wiki/Commands)

rt400, 2018-05-07:
@frawau did you flash tasmota to that board ?

sabag, 2018-05-07:
@rt400 , which board are you referring ? if it is the LX-wifi-03o, than you can read above that several of us were able to flash it. just follow the suggestions outlined above. if you have specific questions, fire away.

frawau, 2018-05-07:
@rt400 Was busy. Did not have time yet.

frawau, 2018-05-07:
@j2mc "For SonOTA you don't need to enter flash mode, just follow their instructions(use the python version, the exe didn't work for me). "   Which instruction are you referring to? I already tried the OTA upgrade and I am now stuck in the "last stage".  So I will need a way to flash it ...

rt400, 2018-05-07:
the board that @frawau have its just what i have..
I managed to flash it.
A power and earth connection already exists (it is on the lower right side, it has a drawing of eight)
The difficulty was finding GPIO 0 and TX RX.
But I found someone who brought up a picture of the Chip and he signaled each leg to what it belonged to.
[ESP8266_SOC2.pdf](https://github.com/arendst/Sonoff-Tasmota/files/1979449/ESP8266_SOC2.pdf)

GPIO 0 solder with a small touch. I turned on the card without a TX RX connection and with two thin wires I pressed on the legs of the chip and a friend of mine pressed the computer on the software and it succeeded. That's the only way I managed to flash firmware.
If anyone knows where there is another TX TX location on this board I'd love to know

closed 2018-07-06 by stale[bot]

jakobeissler, 2020-01-19:
@sabag did you ever manage to fix the led color issue when lights are turned on? I'm having exact same problem as you described.

sabag, 2020-01-20:
@jakobeissler , yes i did, i don't remember doing something special, i just used the commands listed here, ledstate and setoption.

jakobeissler, 2020-01-24:
@sabag i totally missed that, it is working! The difference is that i flashed the device with [tuya convert](https://github.com/ct-Open-Source/tuya-convert).