arendst/Tasmota #1816: Sonoff(WeMo) + GHM ?

marcinbauer85, 2018-02-07:
Hi!
Is there a way to controll a Sonoff /w Tasmota with a Google Home Mini speaker, but NOT with a IFTTT + Web Server combo?

I thought that I can set Sonoff emulation to WeMo (if this setting tries to mimic a WeMo plug) and add a 'WeMo' device in the Google Home app -> Home Devices -> WeMo app, typein my WIFI NAME and Sonoff MAC ADDRESS but in the app I get this:

[image]
[image]

domdomegg, 2018-02-09:
Only Alexa supports wemo directly - and that functionality is built in, not using the wemo smart home skill. The wemo emulation emulates the protocol enough to be detected and controlled by Alexa, but not enough to be controlled across the internet through Belkin's servers, and thus by Google Home.

On the feedback tab of the Google Assistant app, tell them to add emulated wemo or better yet MQTT support so hopefully we can control our devices directly in the future. However, for the mean time the solution you don't like ([IFTTT + web server](http://tinkerman.cat/using-google-assistant-control-your-esp8266-devices/)) is likely the best one.

You may want to check out #999.

marcinbauer85, 2018-02-09:
Thank you for a great answer! Too bad there is no solution... Could Google Actions be an answer to controll an MQTT device?

domdomegg, 2018-02-09:
Annoyingly not easily, but it definitely is possible. You could either write your own [smart home Google Action with something like hook.io](https://github.com/krvarma/sonoff-mqtt) or could use [home assistant as a bridge](https://home-assistant.io/blog/2017/03/28/http-to-mqtt-bridge/), or could run [node-red on a pi with IFTTT](https://www.npmjs.com/package/node-red-contrib-google-action).

joealmonte, 2018-04-25:
It is now possible. Follow the link to BurnsHA instructional YouTube video.
1)  https://www.youtube.com/watch?v=Xk0CafW4_Pg
2)  https://www.youtube.com/watch?v=iS_88VzrFuw

The second link is the updated method, but please take a moment to watch the first link so you have a basic understanding on how it used to work before the updated method.

closed 2018-06-24 by stale[bot]