arendst/Tasmota #1819: Backlog and NTPServer1 issue

reloxx13, 2018-02-07:
Heya,

is it correct that the device is restarting instant if i change the NtpServer by console?

this also stops the backlog. if ntpserver1 is in the middle, the rest wont get saved/changed cuz the device restarted.

` Backlog Hostname Steckdose-1; IPAddress1 0.0.0.0; IPAddress2 192.168.178.1; IPAddress3 255.255.255.0; IPAddress4 192.168.178.1; NtpServer1 0.de.pool.ntp.org; AP 1; SSId1 YYY; SSId2 AAA; WifiConfig 4;`

reloxx13, 2018-02-07:
Okay, i just checked wiki again. The restart is normal, but would it be possible to wait for the backlog done before restart?

reloxx13, 2018-02-07:
Checked with 5.11.1e, device is restarting AFTER backlog is done and everything got saved.

`Backlog Hostname Steckdose-1; IPAddress2 192.168.178.1; IPAddress3 255.255.255.0; IPAddress4 192.168.178.1; NtpServer1 0.de.pool.ntp.org; AP 1; SSId1 xxx; SSId2 ccc; Password2 yyy; WifiConfig 4;`

arendst, 2018-02-08:
Checking...

reloxx13, 2018-02-08:
thanks 👍 will it test it in the evening :)

closed 2018-02-08 by reloxx13