arendst/Tasmota #1821: New bulb type similar to AIthinker named Lohas 9W E27

lnxbil, 2018-02-07:
Hi,

I ordered a bulb named Lohas 9W E27 (810lm, multicolored and dimmable) and looked inside and found another ESP8266 and the necessary pins. The idea was to have a more powerful Sonoff B1 version. It was heavily glued and I needed much more force to open it than any other tasmota-able bulb before:

[image]

After successfully soldering wires to the pins, fashing worked on the first try and it "kind-a" works with the AI-light driver. It lit up and it can be dimmed but to a color instead of anything else. As I found not really much on the bulb itself, it looks very similar to the AI-light, yet different. It as two MY9231 chips, not the MY9291 ones and the PCB is labeled `TYLD1R`, yielding only one electronic-specify link on Google pointing to some page on alipay with a PCB similar to the one in the bulb, yet the ESP stuff is totally different.

The Sonoff B1, which uses the MY9231 according to the source code does not work at all, after playing around with the AI-lIght dimmer settings and changing bulb types, I now have only color, no white at all.

What can I do to help to get it to work?

Lhonco, 2018-04-19:
Did you find a solution to your problems and do you maybe have some tips on how you flashed your Lohas lamp?
I’ve ordered the slightly bigger 10W version from Lohas but I don’t seem to be able to flash it at all. When I try to read or erase the flash with esptool it doesn’t connect, I think the chip is not in flash mode, but connecting GPIO0 to ground isn’t changing anything. I’ve successfully flashed some Sonoff switches etc. with the same connection layout, so my FTDI connector and the drivers on my Computer (MacBook) are working correctly.

MechanicalMink, 2018-05-20:
@Lhonco I'm having the exact same issue as you. Easily able to flash other ESP devices with my computer, FTDI cable, and other hardware, but my bulb (9w from Lohas, no version, TYLD1R board) refuses to enter flash mode when I connect GPIO0 to ground. I'm using the Arduino IDE to program my devices, and it behaves just as if there is nothing connected to the other end of the FTDI cable.

closed 2018-07-21 by stale[bot]

darkfiberiru, 2019-01-12:
Also verified that I'm only able flash these bulb when there using mains power.

DavinKD, 2019-02-12:
Thanks to all who contributed to this thread.  I was able to flash these bulbs OTA using tuya-convert.  

https://smile.amazon.com/gp/product/B07544GPWR/ref=oh_aui_search_asin_title?ie=UTF8&psc=1

https://github.com/ct-Open-Source/tuya-convert

Got the bulb working with slight modifications to 
xdrv_04_light.ino - Modify channels for red, green, blue and white

Set the bulb to AiLight and profit.

ZericE, 2019-03-02:
@DavinKD 
> > Got the bulb working with slight modifications to
> xdrv_04_light.ino - Modify channels for red, green, blue and white
> 
> Set the bulb to AiLight and profit.
What mod did you do to xdrv_04_light.ino? I have the 9w lohas which I also flashed using tuya-convert.

I'm hoping by using the new "template" feature of tasmota, re-building from source won't be required for the lohas (and similar) bulbs. I haven't had a chance to play with recent development builds that include this feature, but plan to.

ZericE, 2019-03-04:
By using the template function of version 6.4.1.19, I was able to get the LoHas RGB 9w bulb mostly working, except for the color blue. Due to the way the channels are arranged for the two my9231 driver ICs, there is no way to get the blue LEDs to illuminate without changing the tasmota code.

I ended up making a couple of small code changes to the sonoff B1 bulb in 6.4.1, and made a special FW that completely works with this bulb. One needs to change the GPIO pins in template.h for the sonoff b1 to use GPIO 13 and 15, and change the order of the channels in xdrv_04_light.ino, function LightMy92x1Duty, from { duty_w, duty_c, 0, duty_g, duty_r, duty_b } to { duty_r, duty_g, duty_b, duty_w, duty_c, 0 } .

It would be rather easy to add support directly for that bulb since it's very similar to the sonoff b1, except for pins and color channels.

tungmeister, 2019-06-12:
@ZericE could you share the firmware please?

rctableoverload, 2019-10-05:
"New" Revision of the LOHAS 9W E27 Bulbs?
ordered at Amazon Germany 01.10.2019 [https://www.amazon.de/gp/product/B07WV7TVBF/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&psc=1](url)

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

djmyst, 2019-10-17:
@ZericE could you please share your firmware as im struggling to get mine to work thanks would be so grateful

Vendo232, 2019-10-28:
Just flashed tasmota 6.7.1 and used this template and the bulb is not working . HELP

{"NAME":"Lohas RGBW","GPIO":[0,0,0,0,0,0,0,0,0,143,0,144,0],"FLAG":0,"BASE":18}

darkfiberiru, 2019-12-06:
Just for reference
Someone figured out how to safely power lohas bulbs to flash them https://majenko.co.uk/blog/more-esp8266-hacking-lohas-led

DavinKD, 2019-12-12:
If anyone is interested, the bin for the modification I made to Tasmota to get these working is here.

https://github.com/DavinKD/SmartThings/blob/master/Lohas1.bin

It's an older version of Tasmota, but it works.

DavinKD, 2019-12-13:
For anyone who stumbles on to this later, the latest tasmota supports this bulb without custom code changes.  Template is {"NAME":"Lohas RGBW","GPIO":[255,255,255,255,255,255,255,255,255,143,255,144,255],"FLAG":15,"BASE":18}