arendst/Tasmota #1822: Lyasi/Meamor bulbs

Jolanrensen, 2018-02-07:
So I bought a Lyasi bulb on Amazon since it also had an ESP8266 chip to try if I could get Tasmota to run on it. However when opening it I was greeted by the lack of Rx/Tx etc contact points.
[Photo album](https://photos.app.goo.gl/AA7OtqQ10U5CEupP2)
As you can see only the (gp)io0, (gp)io2 and RET (whatever that might be, Rest? Rx?) are available. Others might be as well where the green board is soldered to the blue one, but this is unlabled so I don't know. Is it possible to flash Tasmota using only those ports? And if so, how? Or does anyone know the layout of the board?
The bulbs are btw based on the Tuya protocol out of the box.

reloxx13, 2018-02-08:
google esp8266 pins
solder cables to the vcc,rx,rx,gnd pins directly on the chip
flash

Jolanrensen, 2018-02-08:
I would do that, however as you can see in the pictures the Rx/Tx chips on the top right of the chip are behind that cylinder thing. I can't properly see it, let alone solder a cable to it...
I did however get a response from Lyasi themselves and they probably will send me the board schematics (after the Chinese new year)

samuscherer, 2018-03-15:
Hey, did you receive the schematics? If so, do you think it's possible to upload them so we all can have a look at how the bulb is build? Thanks!

helmar74, 2018-03-30:
Depending on your Picture, I think I have the same bulb:
[https://github.com/arendst/Sonoff-Tasmota/issues/2293]

It's running Tasmota and I controll white LED, but not the RGB.
Good informations can be found:
[https://community.home-assistant.io/t/cheap-uk-wifi-bulbs-with-tasmota-teardown-help-tywe3s/40508/39]

Jolanrensen, 2018-04-13:
I think the AI light is slightly different as it does have labeled pins on the top, right? Mine almost has no labeled pins, so I asked Lyasi and got a specsheet from them. I'll try to flash tasmota on it in a week or so. Any idea how to backup the current software btw?

Jolanrensen, 2018-04-23:
Using esptool and the latest sonoff.bin I managed to get the server running! However, the lamp won't stop blinking, which makes sense as it's an unknown device

samuscherer, 2018-04-23:
Sounds like you fried your chip 😆 Do you think you'll get a new one or just buy a non-lyasi bulb?

helmar74, 2018-09-13:
Maybe this helps:
https://community.home-assistant.io/t/cheap-uk-wifi-bulbs-with-tasmota-teardown-help-tywe3s/40508/39

ascillato, 2018-09-13:
Have you make it work?

Jolanrensen, 2018-09-15:
Personally I stopped trying as Home assistant now has native support for Tuya lights, so I just use it that way. But feel free to try yourself as the bulbs aren't that expensive

ascillato2, 2018-09-15:
Hi,

As nobody is working in a driver for this and now this lights are being supported directly by HA softwares, this issue is being closed.

Please, If someone is interested in working on a driver, please, ask to reopen. Thanks.

blawford, 2018-11-15:
So I have just flashed Tasmota to one of these bulbs. I can access the web interface but having no luck getting any light out of the bulb.

The spec says that the LED channels are: 
> IO12 (R), IO15 (G), IO13 (B), IO4 (W) 

I have set the module type to '18 Generic' and set the channel configuration as follows:
> GIO12 = 37 PWM 1
> GIO15 = 38 PWM 2
> GIO13 = 39 PWM 3 
> GIO4 = 40 PWM 4

Having had no luck there I tried setting each individual pin to '40 PWM 4' one at a time and restarting to see whether it would turn on white, but nothing. 

Bulb is reporting status on MQTT but no light. Anyone any idea on how I configure it?

blawford, 2018-11-16:
@ascillato - Could this be reopened? I would be interested in working on this further.

andrethomas2, 2018-11-16:
@blawford Enlighten us ;)

blawford, 2018-11-16:
> @blawford Enlighten us ;)

Haha, I didn't say I had any clue as to what the problem was!

I have got a bit further than the OP, he also thought that the WiFi may have been handled off-chip and not supported by the Tasmota firmware. I haven't found this to be the case as I can access the Tasmota web interface on my device after flashing. 

I am therefore hoping that it might just be a case of configuring the module correctly, rather than having to write any new code (although I am willing to give that a try if need be). I have detailed 3 posts up what I have tried and was wondering if anyone could point me in the right direction of what to do next? I am guessing that the LED channels are driven by another chip on the board? Otherwise I would have seen some life out of the bulb with what I have tried already? Is that situation workable from what is already included in the firmware?

The (translated) spec for the chip is [here ](https://docs.google.com/document/d/1FIXCO-kCga6BeqwOLaMbIiBCxxZkVELyWttfyrsLCR4/edit?usp=sharing), but I am not seeing anything in there I can use to help me configure the module, but I don't have much experience in reading such things, could someone give it a look?

Many thanks!

blawford, 2018-11-16:
Will see what I can do.

Did you have a look at the spec pdf? Does it help at all?

Didn't see anything too helpful on there.

chaosmaster, 2018-11-16:
@blawford 
The OP mentions it uses the tuya protocol.
Have you tried selecting the "54 Tuya Dimmer" in module configuration?
If it indeed uses the tuya protocol it should probably at least allow switching on and off and controlling one channel.
_Weblog 4_ should also give some info.

chaosmaster, 2018-11-16:
I was under the impression, this is a different Bulb (it doesn't use a TYWE3S).
Although it still might use the same implementation.
Does this one have a SM16726/SM16716 as the other one apparently does?

@ascillato native support means using the manufacturers cloud?
I wouldn't really consider this an alternative.

blawford, 2018-11-16:
> @blawford
> The OP mentions it uses the tuya protocol.
> Have you tried selecting the "54 Tuya Dimmer" in module configuration?
> If it indeed uses the tuya protocol it should probably at least allow switching on and off and controlling one channel.
> _Weblog 4_ should also give some info.

Definitely had it fully working with the Tuya app before I flashed it. I have just updated OTA to 6.3 but still nothing after choosing '54 Tuya Dimmer' with the same pin configuration as I mentioned about.

andrethomas, 2018-11-16:
I think I saw some Tuya stuff added to the development codebase which was not included in the 6.3.0 release.

Have you tried the pre-compiled development binary from http://thehackbox.org/tasmota/sonoff.bin

blawford, 2018-11-16:
> I think I saw some Tuya stuff added to the development codebase which was not included in the 6.3.0 release.
> 
> Have you tried the pre-compiled development binary from http://thehackbox.org/tasmota/sonoff.bin

Tried it after you mentioned it, not sure how I should be configuring it but tried the Tuya Dimmer and Generic modules, no luck either way.

andrethomas, 2018-11-17:
Definitely ESP8266 as can be seen in the photos

[image]

As you can see there is a air gap in the PC board which seperates the AC (High voltage) side from the DC (Low voltage side)

The pins that are being used to control the device is soldered from the seperate pc board on which the ESP8266 chip is mounted - I marked it bottom left in above image.

ESP8266 chip with the SPI flash chip is clearly visible here:

[image]

Given the part number of the SPI Flash it is 1M (which is to be expected)

This guy here, circled in red, is what you identified as part number MY9291QD is the driver chip for the LED's - Manufacturer link: http://www.my-semi.com.tw/content/products/product_mean.aspx?id=6 and a not so complete datasheet to be found here: http://www.my-semi.com.tw/file/MY9291_BF_0.91.pdf

[image]

So basically you need to do the following:
1. Figure out which pins of the ESP8266 are brought down to the main board via the soldered join posted in my first image.
2. Figure out how they are connected to the MY9291QD chip.
3. Research if you can find anything else using a ESP8266 and this chip from a circuit and/or code perspective to get insight as to how to interface with it.

I think depending on the outcome of point 3 there is a good chance this one can be made to run Tasmota.

andrethomas, 2018-11-17:
[image]

So there will be data comming from the ESP8266 on only two pins for DI (data) and DCKI (data clock input)

They would not have used normal serial - this chip will have some two wire protocol that it uses.

I marked the pins of interest with BLUE circles on the diagram, and blue lines on the chip footprint itself. Those pins should match up to two of the soldered connection to the ESP8266 board. The pushbutton visible in the previous posts last image is most likely connected to GPIO0 so there is a good change that (VCC, GND, GPIO0, RX, TX) and (DI,DCKI) will be among those 8 solder joins from the ESP8266 board to the main board.

If you have a multimeter with diode-test (that function that beeps when you connect the two probes to test for continueity) you should be able to map them out with some luck.

andrethomas, 2018-11-17:
Links of potential interest:
http://supersimo88.altervista.org/ailight-an-esp8266-rgbw-light-bulb/

And here is what appears to be a arduino driver library for the LED controller chip
https://github.com/xoseperez/my9291

arendst, 2018-11-17:
My9291 is being used by both Tasmota AiLight and Sonoff B1 modules!

Did you try the AiLight or Sonoff B1 module?

blawford, 2018-11-17:
> My9291 is being used by both Tasmota AiLight and Sonoff B1 modules!
> 
> Did you try the AiLight or Sonoff B1 module?

Yes, I found the same doing some Googling the MY9291. No I haven't tried either of them.

I will attempt to but the bulb back together and give it a try, not entirely sure how I go about reconnecting to the threaded part of the E27 connector but I'm sure I will work it out.

arendst, 2018-11-17:
Before putting it all together perhaps you can manage to find out the correct GPIO connections to the My9291 chip ;-)

The interface only uses two GPIOs.
AiLight uses GPIO13 for DI and GPIO15 for DCK
Sonoff B1 uses GPIO12 for DI and GPIO14 for DCK

You might be able to measure no resistance between pins 15/16 of the My9291 chip to any of the GPIOs of the ESP8266.

andrethomas, 2018-11-22:
@blawford Did you make any further progress?

blawford, 2018-11-22:
Unfortunately not, upon trying to get enough access to solder where I needed for reassembly, long story short, the bulb is never going back together!

With that said, I was assuming I should just be able to connect the red/black leads of the bulb to live/neutral on a cable with a plug on the end and have it work as if it was connected in the lamp holder? I did try this briefly and saw no signs of life from the bulb, so it is possibly bricked.

I would like to get it up and running so this hasn't all been for nothing! For the sake of others if not my own. Might consider buying another bulb too as access for soldering wasn't too difficult. If I am unable to verify this as working I would probably give up on it and just buy a Sonoff B1.

andrethomas, 2018-11-23:
Its not possible to brick the ESP chip, if that is what you mean by bricked.

Also, did you by chance manage to check the pin map with a multi-meter as recommended by @arendst ?

It quite possibly is compatible with either the AiLight or the B1.

blawford, 2018-11-23:
> Its not possible to brick the ESP chip, if that is what you mean by bricked.

I had to remove a capacitor to get to the pins on one of the chips, so it could have been something around doing that which has caused the problem.

Tried to check the pins with a multimeter, I wouldn't be able to say with enough certainty what was connected to what, the chips are so small! I possibly have a rough enough idea that I might be able to try the combinations if I can get the bulb up and running again though.

andrethomas, 2018-11-23:
I have a modified set of leads for my multimeter that has normal sewing pins soldered to the end so that I can test in high density areas.

These kind of pins:
[image]

arendst, 2018-11-23:
But how do you fix these pins to the device?

 

I have “pogopins” also soldered to wires but still have to find an easy way to connect them to the device. Currently they are just present; never used.

 

https://www.banggood.com/nl/100pcs-P75-E2-Spring-Test-Probe-Pogo-Pin-Dia_-1_02mm-Pin-Tool-p-1247554.html?gmcCountry=NL&currency=EUR&createTmp=1&utm_source=googleshopping&utm_medium=cpc_union&utm_content=2zou&utm_campaign=pla-all2-nl-dutch-pc&gclid=Cj0KCQiA597fBRCzARIsAHWby0HT4ksCenLUKidZKScnxfdra76Y_pFjXNkkJUl8wbU0PHTnvYpkToUaAsrgEALw_wcB&ID=41299&cur_warehouse=CN

 

 <https://user-images.githubusercontent.com/470015/48937078-b253fb00-ef15-11e8-992a-190ee7ae690f.png>

andrethomas, 2018-11-23:
Yeah, it is just soldered onto the end of the multi meter leads to have super thin probes that can test pins and pcb traces down to 0.4mm pitch no problem.

ascillato, 2018-11-27:
@blawford 

Hi,

Any news on this? There is anything where we can help?

ascillato2, 2018-12-03:
@blawford

Hi,

Any news on this? There is anything where we can help?

Eichhoernchen, 2019-01-28:
I got the same bulb.
I just opened it up and removed the capacitor to get access to the ESP and the my9291.
It seems that my9291 DCKI maps to PIN 24 (GPIO 5) and my9291 DI maps to PIN 16 (GPIO 4).
When looking at the image posted before where the ESP pins are highlighted in yellow, these are the pins to the top right corner with DCKI being the rightmost.

Since it is the first time for me to fiddle with the tasmota firmware, can anyone hint me at how to get this into the firmware? Is the sonoff_template.h the right starting place (for the beginning I just wanted to modify the B1 oder the AiLight). How are the pins numbered within the tasmota firmware?
Do I need to simply move the GPIO_DI and GPIO_DCKI to the GPIO 4 and 5 position in the file?

Eichhoernchen, 2019-01-28:
So, short update, I got the stuff compiled.
The light works flawlessly with the AiLight on GPIO 4 and GPIO 5, the B1 does not work properly, so it seems the bulb is a `LT_RGBW`.

Eichhoernchen, 2019-01-29:
So,
I would be happy to contribute this, but there are two solutions to this.

A: The strawman solution is kind of copying the AiLight module and creating a new one with the modified pin assignment.

B: The, in my opinion, better solution would be to enable the selection of the bulb type and DI and DCKI pin assignment via the web interface, this would allow to collapse all my9192-based bulbs into a single module. However, I'm not really familiar with the firmware and how one would do this, from quickly looking at some files it seems that such a design is not intended.

So, I'm happy to contribute solution A. So please let me know if I should file a PR.

Apart form that I noticed that the colors are a bit off compared to the stock firmware.
I saw that there seems to be some sort of color correction via `ledTable`, how is that table generated, is that somehow dumped from the original firmware?

victorChB, 2019-05-15:
Is this still 

> So,
> I would be happy to contribute this, but there are two solutions to this.
> 
> A: The strawman solution is kind of copying the AiLight module and creating a new one with the modified pin assignment.
> 
> B: The, in my opinion, better solution would be to enable the selection of the bulb type and DI and DCKI pin assignment via the web interface, this would allow to collapse all my9192-based bulbs into a single module. However, I'm not really familiar with the firmware and how one would do this, from quickly looking at some files it seems that such a design is not intended.
> 
> So, I'm happy to contribute solution A. So please let me know if I should file a PR.
> 
> Apart form that I noticed that the colors are a bit off compared to the stock firmware.
> I saw that there seems to be some sort of color correction via `ledTable`, how is that table generated, is that somehow dumped from the original firmware?
> 
> Edit: for reference, here is a change to 6.4.1 adding this as a module: https://pastebin.com/0aWxp9ZQ

Is this still the same for release 6.5.0 ? I mean do I have to modify the pins in the firmware and flash it?
or can I use the web interface, because I see that there are two options for the GPIOs (my92x1 DI (143) my92x1 DCKI (144)) but havent been able to control the Lyasi bulb.