arendst/Tasmota #1823: Issues with re-flashing a TH-16

boilermanc, 2018-02-07:
I flashed a TH-16 and had it connected to my wifi and setup with mqtt and connected to openhab. all good. but the temp was set to Celsius and i needed Fahrenheit. so as i understand you have to make the change from the 0 for Celsius to the 1 for Fahrenheit in the setup file. so i made the change and flashed the sonoff again. completed just fine but did not change the degree. 

so i tried to change the address of the sonoff to see if it was flashing correct (it actually held all of my mqtt setup values that i had put in there so that that was weird.). it would not connect to the new ip address i put in. 

so then i uploaded a script to wipe the sonoff which it did. then i reloaded the script and it still does not connect to another ip, will connect to the old ip (when i change that back in the script), and does not change the degree setting. 

hope that all makes sense...

thanks!

Frogmore42, 2018-02-07:
First, you don't need to change the temp units setting in the config file, you can do it from the console or MQTT.  Use:
SetOption8 1

There are different ways to do things (configuration) and it is not obvious what is the best way.  In code (not in front of me now) there is a cfg_holder (or something like that) that is a number.  This number gets stored in the flash and if the new code has the same number, the current settings in the flash are retained.  If you change the number, the default settings from the image are loaded.

My new proposed way (I have not actually implemented it yet, but will soon) is to leave the cfg holder alone so that I can upload code easier (and not loose my settings).  Before I upload code the first time, I make sure I have set my network settings and MQTT settings, since these are necessary to talk to the device and the same for all of my devices.  I then build the code, flash it, and then connect to the Web UI and configure the module the way I want it.  Later, when I want to update the firmware, I can just flash the new FW and it will have the same settings it had before.

The way I have been doing it, I have used different numbers for the cfg holder and not keeping good track of the numbers.  So, later when I reflash, there is a high probability the cfg holder will be different and my settings will be lost.  Since I have my network and MQTT settings in there, it is not too bad.  But it does make it much less likely I will update the FW.

boilermanc, 2018-02-14:
Hi Frogmore, thanks! That worked to change the temp. excellent.

as for the config holder, your talking about this line?

#ifndef _USER_CONFIG_H_
#define _USER_CONFIG_H_

just put in say a 1 to change it? then 2, 3, 4 if needed?

#ifndef _USER_CONFIG_H_
#define _USER_CONFIG_H_1

thanks!

Frogmore42, 2018-02-15:
This is the line
#define CFG_HOLDER             0x20161209        // [Reset 1] Change this value to load following default configuration parameters

But, go be clear I don't recommend changing it normally.  Doing that will likely get you into trouble later, when you want to update FW but don't want to reset everything.  If you change the default settings, you can get them to take with a reset command.

boilermanc, 2018-02-15:
Thanks! So... what is the reset command? sorry...

Frogmore42, 2018-02-15:
https://github.com/arendst/Sonoff-Tasmota/wiki/Commands
Reset 1
Reset 2
You can send it using console, mqtt, or web commands.