arendst/Tasmota #1824: Strange Intermittent WiFi Connectivity

ChaoticConundrum, 2018-02-07:
I'm having an issue with WiFi connectivity. It is not likely to be signal strength, since the access point this is connecting to is three feet from the Sonoff Basic I'm testing. I'm uploading firmware with the Arduino IDE over serial, and setting WiFi credentials in user_config.h, which looks like:
```
// -- Wifi ----------------------------------------
#define WIFI_IP_ADDRESS        "0.0.0.0"         // [IpAddress1] Set to 0.0.0.0 for using DHCP or IP address
#define WIFI_GATEWAY           "10.0.0.1"        // [IpAddress2] If not using DHCP set Gateway IP address
#define WIFI_SUBNETMASK        "255.255.240.0"   // [IpAddress3] If not using DHCP set Network mask
#define WIFI_DNS               "192.168.2.27"    // [IpAddress4] If not using DHCP set DNS IP address (might be equal to WIFI_GATEWAY)

#define STA_SSID1              "WatersExt"          // [Ssid1] Wifi SSID
#define STA_PASS1              "<redacted>"        // [Password1] Wifi password
#define STA_SSID2              ""                  // [Ssid2] Optional alternate AP Wifi SSID
#define STA_PASS2              ""                // [Password2] Optional alternate AP Wifi password
#define WIFI_CONFIG_TOOL       WIFI_RETRY    // [WifiConfig] Default tool if wifi fails to connect
                                                 //   (WIFI_RESTART, WIFI_SMARTCONFIG, WIFI_MANAGER, WIFI_WPSCONFIG, WIFI_RETRY, WIFI_WAIT)
```
Also, `SERIAL_LOG_LEVEL` is `LOG_LEVEL_ALL`.

What is confusing to me is on a reset, every time, the Sonoff briefly connects to the MQTT server and manages to publish a LWT = Online. Sometimes it publishes a few other values, but they usually don't make it to the server. After a few seconds, the server publishes a LWT = Offline (it lost connection with the Sonoff).
```
00:00:00 APP: Boot Count 1
00:00:00 Project sonoff1 Sonoff (Topic sonoff1, Fallback DVES_80C037, GroupTopic sonoffs) Version 5.11.1
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP1 WatersExt in mode 11N as sonoff1-0055...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Checking connection...
00:00:03 WIF: Attempting connection...
00:00:04 WIF: Checking connection...
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Checking connection...
00:00:05 WIF: Connected
00:00:05 HTP: Web server active on sonoff1-0055 with IP address 10.0.1.35
00:00:07 MQT: Attempting connection...
00:00:07 MQT: Connected
00:00:07 MQT: hab/sonoff1/telem/LWT = Online (retained)
00:00:07 MQT: hab/sonoff1/cmd/POWER =
00:00:07 MQT: Subscribe to hab/sonoff1/cmd/#
00:00:07 MQT: Subscribe to hab/sonoffs/cmd/#
00:00:07 MQT: Subscribe to cmnd/DVES_80C037/#
00:00:07 MQT: hab/sonoff1/telem/INFO1 = {"Module":"Sonoff Basic","Version":"5.11.1","FallbackTopic":"DVES_80C037","GroupTopic":"sonoffs"}
00:00:07 MQT: hab/sonoff1/telem/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff1-0055","IPAddress":"10.0.1.35"}
00:00:07 MQT: hab/sonoff1/telem/INFO3 = {"RestartReason":"Software/System restart"}
00:00:08 MQT: hab/sonoff1/stat/RESULT = {"POWER":"OFF"}
00:00:08 MQT: hab/sonoff1/stat/POWER = OFF
00:00:15 RSL: hab/sonoff1/telem/STATE = {"Time":"1970-01-01T00:00:15","Uptime":0,"Vcc":3.475,"POWER":"OFF","Wifi":{"AP":1,"SSId":"WatersExt","RSSI":100,"APMac":"22:4E:7F:6C:8B:1A"}}
00:00:25 WIF: Checking connection...
00:00:25 WIF: Connect failed as AP cannot be reached
00:00:25 WIF: Connecting to AP1 WatersExt in mode 11N as sonoff1-0055...
00:00:26 WIF: Checking connection...
00:00:26 WIF: Attempting connection...
```
And then it continues indefinitely until I restart it (either remove power or hold the button down), then I see the behavior above and repeat.
```
00:00:56 WIF: Checking connection...
00:00:56 WIF: Attempting connection...
00:00:57 WIF: Checking connection...
00:00:57 WIF: Connect failed with AP timeout
00:00:57 WIF: Connecting to AP1 WatersExt in mode 11N as sonoff1-0055...
00:00:58 WIF: Checking connection...
00:00:58 WIF: Attempting connection...
00:00:59 WIF: Checking connection...
00:00:59 WIF: Attempting connection...
00:01:00 WIF: Checking connection...
00:01:00 WIF: Attempting connection...
00:01:01 WIF: Checking connection...
00:01:01 WIF: Attempting connection...
00:01:02 WIF: Checking connection...
00:01:02 WIF: Attempting connection...
00:01:03 WIF: Checking connection...
00:01:03 WIF: Attempting connection...
00:01:04 WIF: Checking connection...
00:01:04 WIF: Attempting connection...
00:01:05 WIF: Checking connection...
00:01:05 WIF: Connect failed as AP cannot be reached
00:01:06 WIF: Checking connection...
00:01:06 WIF: Connect failed as AP cannot be reached
00:01:06 WIF: Connecting to AP1 WatersExt in mode 11N as sonoff1-0055...
00:01:07 WIF: Checking connection...
00:01:07 WIF: Attempting connection...
00:01:08 WIF: Checking connection...
00:01:08 WIF: Attempting connection...
```
I'd love to buy a bunch of these, and It'd be okay if the only issue was WiFi signal strength, but as said above, the AP is basically point blank.

Let me know if there is any more information I can give, or debugging steps/tips, or even where I should look in the code to get more information about why the Sonoff is disconnecting. Thanks.

reloxx13, 2018-02-08:
Try to set both AP (SSID 1 and SSID 2 and passwords) to the same wifi.

I also noticed that after a restart, AP2 is the first choice instead of AP1.

ChaoticConundrum, 2018-02-09:
Unfortunately, I have tried that, with similar results. It switches between trying the 2 APs, but they still repeatedly fail to connect.

savage872, 2018-02-15:
Have you tried setting static IP settings?

from your log ""Wifi":{"AP":1,"SSId":"WatersExt","RSSI":100,"
RSSI is signal strengh and it's 100 which means you can't get any better.

ChaoticConundrum, 2018-02-16:
Thanks for the suggestion. Unfortunately, I get almost the same results with a static IP. 

```
// -- Wifi ----------------------------------------
#define WIFI_IP_ADDRESS        "10.0.2.100"         // [IpAddress1] Set to 0.0.0.0 for using DHCP or IP address
#define WIFI_GATEWAY           "10.0.0.1"        // [IpAddress2] If not using DHCP set Gateway IP address
#define WIFI_SUBNETMASK        "255.255.240.0"   // [IpAddress3] If not using DHCP set Network mask
#define WIFI_DNS               "75.75.75.75"    // [IpAddress4] If not using DHCP set DNS IP address (might be equal to WIFI_GATEWAY)

#define STA_SSID1              "WatersExt"          // [Ssid1] Wifi SSID
#define STA_PASS1              "<redacted>"        // [Password1] Wifi password
#define STA_SSID2              ""      // [Ssid2] Optional alternate AP Wifi SSID
#define STA_PASS2              ""  // [Password2] Optional alternate AP Wifi password
#define WIFI_CONFIG_TOOL       WIFI_RETRY    // [WifiConfig] Default tool if wifi fails to connect
                                                 //   (WIFI_RESTART, WIFI_SMARTCONFIG, WIFI_MANAGER, WIFI_WPSCONFIG, WIFI_RETRY, WIFI_WAIT)
```

```
00:00:00 APP: Boot Count 1
00:00:00 Project sonoff1 Sonoff (Topic sonoff1, Fallback DVES_80C037, GroupTopic sonoffs) Version 5.11.1
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP1 WatersExt in mode 11N as sonoff1-0055...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Checking connection...
00:00:03 WIF: Connected
00:00:03 HTP: Web server active on sonoff1-0055 with IP address 10.0.2.100
00:00:05 MQT: Attempting connection...
00:00:05 MQT: Connected
00:00:05 MQT: hab/sonoff1/telem/LWT = Online (retained)
00:00:05 MQT: hab/sonoff1/cmd/POWER =
00:00:05 MQT: Subscribe to hab/sonoff1/cmd/#
00:00:05 MQT: Subscribe to hab/sonoffs/cmd/#
00:00:05 MQT: Subscribe to cmnd/DVES_80C037/#
00:00:05 MQT: hab/sonoff1/telem/INFO1 = {"Module":"Sonoff Basic","Version":"5.11.1","FallbackTopic":"DVES_80C037","GroupTopic":"sonoffs"}
00:00:05 MQT: hab/sonoff1/telem/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff1-0055","IPAddress":"10.0.2.100"}
00:00:05 MQT: hab/sonoff1/telem/INFO3 = {"RestartReason":"Software/System restart"}
00:00:06 MQT: hab/sonoff1/stat/RESULT = {"POWER":"OFF"}
00:00:06 MQT: hab/sonoff1/stat/POWER = OFF
00:00:13 RSL: hab/sonoff1/telem/STATE = {"Time":"1970-01-01T00:00:13","Uptime":0,"Vcc":3.486,"POWER":"OFF","Wifi":{"AP":1,"SSId":"WatersExt","RSSI":100,"APMac":"22:4E:7F:6C:8B:1A"}}
00:00:23 WIF: Checking connection...
00:00:23 WIF: Attempting connection...
00:00:23 WIF: Connecting to AP1 WatersExt in mode 11N as sonoff1-0055...
00:00:24 WIF: Checking connection...
00:00:24 WIF: Attempting connection...
00:00:25 WIF: Checking connection...
00:00:25 WIF: Attempting connection...
```
However, it seems to occasionally connect for a few seconds, then times out again.
```
00:03:39 WIF: Checking connection...
00:03:39 WIF: Attempting connection...
00:03:40 MQT: Attempting connection...
00:03:40 MQT: Connected
00:03:40 MQT: hab/sonoff1/telem/LWT = Online (retained)
00:03:40 MQT: hab/sonoff1/cmd/POWER =
00:03:40 MQT: Subscribe to hab/sonoff1/cmd/#
00:03:40 MQT: Subscribe to hab/sonoffs/cmd/#
00:03:40 MQT: Subscribe to cmnd/DVES_80C037/#
00:03:40 WIF: Checking connection...
00:03:40 WIF: Connected
00:04:00 WIF: Checking connection...
00:04:00 WIF: Attempting connection...
00:04:00 WIF: Connecting to AP1 WatersExt in mode 11N as sonoff1-0055...
00:04:01 WIF: Checking connection...
00:04:01 WIF: Attempting connection...
```

ChaoticConundrum, 2018-02-16:
Note I've also tried with a static IP and both APs configures in the user_config.h, with the same results.

savage872, 2018-02-16:
this is what is have on my config

// -- Wifi ----------------------------------------
#define WIFI_IP_ADDRESS        "192.168.0.103"         // [IpAddress1] Set to 0.0.0.0 for using DHCP or IP address
#define WIFI_GATEWAY           "192.168.0.1"   // [IpAddress2] If not using DHCP set Gateway IP address
#define WIFI_SUBNETMASK        "255.255.255.0"   // [IpAddress3] If not using DHCP set Network mask
#define WIFI_DNS               "192.168.0.1"    // [IpAddress4] If not using DHCP set DNS IP address (might be equal to WIFI_GATEWAY)

#define STA_SSID1              "Home Sweet Home"      // [Ssid1] Wifi SSID
#define STA_PASS1              ".........."  // [Password1] Wifi password
#define STA_SSID2              "indebuurt2"      // [Ssid2] Optional alternate AP Wifi SSID
#define STA_PASS2              "VnsqrtnrsddbrN"  // [Password2] Optional alternate AP Wifi password
#define WIFI_CONFIG_TOOL       WIFI_WPSCONFIG    // [WifiConfig] Default tool if wifi fails to connect
                                                 //   

I've seen that you on a larger subnet, try to go for 255.255.255.0 if you can.
is this the only device on your network that you have problems with?

try setting up a hotspot on your phone and see if it's the same or you might just have a bad unit..
hope it helps.

crra, 2018-04-19:
I had the same issue. This worked for me:

1. https://www.reddit.com/r/homeassistant/comments/84p9z8/sonoff_with_tasmota_firmware_wifi_issue/ -> downgrade to 2.3.0
2. #1106 -> `reset 2`