arendst/Tasmota #1826: Web server problems

wayne47git, 2018-02-08:
I posted a lot of info at the bottom of  [this issue](https://github.com/arendst/Sonoff-Tasmota/issues/1681) a few days ago and did not see a response so I'm creating a new issue.

The web server **is** up, it just won't talk to most browsers. Wireshark dumps at the bottom of the prior post.

Makes it really hard to configure. I'm basically stuck at this point and can't figure out how to proceed.

curzon01, 2018-02-08:
I can reproduce the issue with my mobile chrome browser and a fresh compiled "5.11.1i" (commit 150de9c) using esp library 2.4.0 [Core/SDK Version 2_4_0/2.1.0(deb1901)].

It can not be reproduce it using chrome on Linux or Firefox.

The same device with same dev release but using library 2.3.0 [Core/SDK Version 2_3_0/1.5.3(aec24ac9) ] works on all browser.

So the solution might be hint from arendst in the issue you linked above, not using lib 2.4.0 atthe moment and compile it using lib 2.3.0

wayne47git, 2018-02-08:
Thanks for the response. I'm new to this. Got a link/pointer to how to "use lib 2.3.0" and/or "shut off emulation"?

curzon01, 2018-02-08:
### Usage of Lib 2.3.0
For beginners I suggest ready compiled binaries: Go to [releases](https://github.com/arendst/Sonoff-Tasmota/releases), last [5.11.1](https://github.com/arendst/Sonoff-Tasmota/tree/v5.11.1) is already compiled with lib 2.3.0

### Self compiling
Depends how you compile it, see Wiki [Theo's Tasmota Tips](https://github.com/arendst/Sonoff-Tasmota/wiki/Theo's-Tasmota-Tips) - when using Platform IO see hint [20180103](https://github.com/arendst/Sonoff-Tasmota/wiki/Theo's-Tasmota-Tips#20180103---tasmota-is-ready-for-esp8266arduino-version-240):
Edit `platformio.ini` and set
`platform = espressif8266@1.5.0 ; esp8266/Arduino version 2.3.0`

wayne47git, 2018-02-09:
Thank you. I was able to upgrade the firmware using lynx and now it does, indeed, work from Firefox/Chrome.

closed 2018-02-09 by wayne47git