arendst/Tasmota #1827: tasmota 5.11.1 MQTT Payload issues

Caswell1000, 2018-02-08:
1. I flashed Tasmota 5.11.1 (Thanks Arends for the dev work you are doing), I configured MQTT using the available menus, and tested them on the console menu. "Power ON", "Power OFF", "Power 1", etc, all work on the console. 
[image]

2. I installed an app called MQTT dashboard on Android, and sent a topic like this 
/cmnd/Testchip/Power  

and it results in this output (which seems to be correct):

[image]

3. When I send the same topic with a payload (/cmnd/Testchip/Power ON), I get : 

[image]

All topics with payloads, whether numerical (1, 2, 3, ...) or alphabetical ("ON", "OFF", "TOGGLE") result in:

[image]

I am very keen to solve this problem, but my attempts with different combinations and different MQTT broker apps yield the same, please advise how I can change my Topic to make Payloads work.

curzon01, 2018-02-08:
Nobody can help if an issue can not be reproduced. And for reproduce an issue, the tester needs to do the same as you. And you do not real want that a tester try every of the hundred apps with thousand of possible configuration variety.

An android app "

> MQTT Dashboard

" does not exist, it exists "_IOT MQTT Dashboard_", "_Linear MQTT Dashbard_" and several other "*MQTT Dash*" apps. Which of these hundred apps do you are using? And how did you configure it?

**In general**
- the MQTT implementation is essential in tasmota and it works as expected
- do not use leading / in topics. It is allowed but it produce an unnecessary empty topic level with a zero character at front

**Try reproduce**
A simple test using mosquitto and the same Tasmota configuration you re using confirms that the MQTT on Tasmota is working as expected. The problem can only be the Android app you are using but we do not know which app and how (config).

Here the tests with same config as you on Tasmota side but using command line mosquitto_pub for for publishing topics:

```
CMD: Status 6
MQT: /stat/Testchip/STATUS6 = {"StatusMQT":{"MqttHost":"172.17.9.3","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_0EBA54","MqttUser":"mqttdev","MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
CMD: Topic
MQT: /stat/Testchip/RESULT = {"Topic":"Testchip"}
CMD: FullTopic
MQT: /stat/Testchip/RESULT = {"FullTopic":"/%prefix%/%topic%/"}
```

`$ mosquitto_pub -h localhost -u mqttdev -P ****** -t '/cmnd/Testchip/Power' -m ''`

```
MQT: /stat/Testchip/RESULT = {"POWER":"OFF"}
MQT: /stat/Testchip/POWER = OFF
```

`$ mosquitto_pub -h localhost -u mqttdev -P ****** -t '/cmnd/Testchip/Power' -m 'ON'`

```
MQT: /stat/Testchip/RESULT = {"POWER":"ON"}
MQT: /stat/Testchip/POWER = ON
```
`$ mosquitto_pub -h localhost -u mqttdev -P ****** -t '/cmnd/Testchip/Power' -m '1'`

```
MQT: /stat/Testchip/RESULT = {"POWER":"ON"}
MQT: /stat/Testchip/POWER = ON
```

Caswell1000, 2018-02-08:
Thank you, Its IoT MQTT Dashboard, and I am iot.eclipse.org as a broker, the screenshot is shown below:

[image]

The screenshot of my configuration is as follows:

[image]

The app shows that there is connection to the broker

[image]

Is there something wrong with the topic, apart from the leading "/", please assist, I am a noob, but I also read a lot, it could be that I am missing something small, the sonoff is on "Pulsetime 10"

Thanks for taking your time to respond.

arendst, 2018-02-08:
Remove the ``on`` and add it to field ``Value to Publish`` as that is what you want; you want to publish the ``On`` value for topic ``/cmnd/Techchip/power``. Keep reading you're almost there ;-)

Caswell1000, 2018-02-08:
Brilliant, this works perfectly, what's the proper way to subscribe on the iot mqtt bash board app if you want to get the messages from the sonoff in the app, I tried the one below, but I get "n/a" , otherwise thanks a lot for the advise, I'm so excited. Thank you very much. 

[image]

curzon01, 2018-02-08:
@Caswell1000: You always do the same mistake - do not add parameters (here like `-n`) to topics - a subscribe topic `/cmnd/Testchip/status -n` does not exist, there exist for example `/stat/Testchip/STATE` having a JSON object.

Please carefully read [MQTT Topic definition](https://github.com/arendst/Sonoff-Tasmota/wiki/MQTT-Features#mqtt-topic-definition), it's already described.

Caswell1000, 2018-02-08:
Thank you @curzon01 , "/cmnd/Testchip/Power" in the subscription returns the value I have used under value to publish, my hope is to get something that the sonoff publishes, i.e. a direct status message from sonoff, not what I put under value to publish. I actually need to subscribe to a stat message, but "/stat/Testchip/Power" returns n/a so does "/cmnd/Testchip/status", perhaps the subscription is meant to just give you the same value one puts under value to publish. 

In graphics: When The Publication is:

[image]

And the Subscription is:

[image]

the result is :

[image]

but when the subscription is:

[image]

I get : 

[image]

How do I get a status report from the sonoff??

Thanks for all the help that you are providing.

tyjtyj, 2018-02-08:
Try enable json converter

curzon01, 2018-02-08:
/stat/Testchip/POWER

not

/stat/Testchip/power

Caswell1000, 2018-02-09:
Many Thanks, everything is now in order,  /stat/Testchip/POWER worked, I appreciate all the help I've had.

closed 2018-02-09 by Caswell1000

centiane, 2018-06-25:
Hi , I use same app but i not found for me.
This is mi config MQTT in sonoff Tasmota.
[image]

and this is mi Broker
[image]

how activate a power1 on?
I am a novice, and sorry for my english.
Regards Centiane.