arendst/Tasmota #1828: Provide simple way to add own application code

tobox, 2018-02-08:
I have read the articles about using a tasmota device as a thermostat (mainly #481 and #818), and I understand the pros and cons. 

Given the vast number of inputs and outputs supported in tasmota, I cannot come up with a simple solution that would be completely configurable at runtime. 

Long before Tasmota, I used Ethersex on AVR NetIO boards, and they had simple solution for this: there was a "sample application" file which was basically empty, with just some headers for initialisation, a recurring event loop and a function that could be triggered using the command line interface. I think cloning that concept into tasmota and having a wiki page which describes the most common scenarios would be really helpful.

E.g., my parents don't have any home automation and no MQTT server, but they would like to run a dehumidifier if its too damp. For all those applications I would just write a specifically tailored application file, and as long as the "API" to tasmota does not change, I could easily update to newer versions by just recompiling tasmota with my application file.

I think the API should provide at least the following calls:

- init/setup: run once after startup
- loop1s: run every second
- loop0_05s: run every 0.05 seconds (I don't currently need that, but for some applications it might be helpful)
- onCommand(char *arg): to manually trigger things using the command interface
- finish: runs before shutdown/reboot

What do you think about that? Most simple switch functionalities would be only one line in the loop function, but I think the framework works for more complex use cases.

arendst, 2018-02-08:
Lucky for you more than most of it is already available!!!

Take a look at how the sensors are implemented in Tasmota. Scroll all the way down and see the use of function calls. Scroll up a little and see how you append to both JSON message and root webpage.

Take a look at how the devices are implemented too with regard to command recognition.

It's all based on (not well defined) function call. To see all available function calls look in both xdrv_interface.ino and xsns_interface.ino. Tou might want to document these calls to help others out.

Keep persistent data at a minimum to not always have to update flash settings lay-out.

Enjoy!

tobox, 2018-02-08:
Excellent! I have just created a xdrv_08_user_application.ino that does exactly what I wanted. I will add some more documentation and the open a pull request for review.

arendst, 2018-02-08:
To make a difference between released and user development sensors/drivers I suggest to select an index between 91 and 99 instead of your current 08. This feature is is just made available in the latest dev 5.11.1j release.

galak, 2018-02-09:
I've got one of the DLX-KS602S wall switches and want to hook some code to when the switch is pushed (on or off).  Not sure if the driver/sensor hook is what I should be looking at, if so any start points/examples I should look at?

A little more detail - trying to write some custom code to use the DLX-KS602S to control a Haiku fan.  The Haiku fan is connectable over TCP/IP and can turn it on/off with simple commands like:
<Master Fan;LIGHT;PWR;ON>
<Master Fan;LIGHT;PWR;OFF>

tobox, 2018-02-12:
@galak, it is far from complete, but here is my current state of the user application sample code for tasmota:

https://github.com/tobox/Sonoff-Tasmota

Currently, it provides for a periodic event loop and 2 sample API commands, but the connection to sensor is still missing.

colinl, 2018-03-02:
To clarify what this is about, is the idea that I would use the file sonoff/xdrv_98_user_application.ino as a template into which I would insert code to perform the task that I wish to achieve?
On a point of order, should you not be doing this on a new branch in your repo rather than on the development branch?

colinl, 2018-03-03:
@tobox (and @arendst), first to say thanks for doing this, it has enabled me to get up and running with some DIY code very quickly.
One point, after handling a command I am getting an MQTT publish of
stat/RESULT = {"COMMAND":"ERROR"}
to correct this, in User_Application_Command() after handling a command one can add something like

`snprintf_P(mqtt_data, sizeof(mqtt_data), PSTR("{\"" D_CMND_USER_APPLICATION D_CMND_USER_APPLICATION_CMD_A "\":\"%s\"}"),XdrvMailbox.data);`

I am not absolutely certain this is the ideal way of doing it as I am still feeling my way around the code, but it does seem to cause the appropriate result.

I would have raised an issue against tobox's repo but issues do not seem to be enabled there.

colinl, 2018-03-04:
I have a working example going, which generates a time proportioned relay output from a linear power requirement in the range 0:1, with a configurable cycle time. So, for example, if the cycle time is configured to be 60 seconds and the required power (which is passed via mqtt) is set at 0.25 then the relay will be on for 15 seconds and off for 45 seconds in each 60 second period. It can be seen in my timeprop branch at [https://github.com/colinl/Sonoff-Tasmota/tree/timeprop_branch](https://github.com/colinl/Sonoff-Tasmota/tree/timeprop_branch). Just the three files lib/ProcessControl/Timeprop.cpp and Timeprop.h and sonoff/xdrv_91_timeprop.ino are needed.  
Instructions for config and use in xdrv_91_timeprop.ino

I don't know whether there would be any interest in including this feature in the standard system.

colinl, 2018-03-15:
I am trying to understand the operation of Xdrvnn(FUNC_SET_POWER). I can see that for a single relay output then this is called when the relay state changes with the new state in the lsb of XdrvMailbox.index.    I think that for multiple outputs then successive bits will hold the current state of the other relays.  Since I do not have h/w with multiple relays I have not been able to check that.  Can you confirm whether my deductions are correct please?

arendst, 2018-03-15:
If you use visual studio code (VSC) with platformio you can easily search on FUNC_SET_POWER. You then see that your assumptions are correct and that the function is only used by some drivers and not the relay driver ;-).

If developing for Tasmota I strongly suggest to use VSC as it covers all files so much easier than the Arduino IDE.

I only use the Arduino IDE for backward compatibility testing (and currently to compile for 2.4.1 ;))

colinl, 2018-03-15:
@arendst, thanks for confirming what I had deduced.  I use Ubuntu with Atom and platformio and had done exactly what you suggested, which is how I had come to my conclusion.  However, since I will be publishing my code for others to use, and am not familiar in depth with your excellent s/w, I just wanted confirmation of what I thought in order to minimise the likelihood of my untested code failing when others try to use it.
Thanks again
Colin

Stevehans, 2018-04-01:
Does this also provide a convenient way for a user app to extend the web interface?

martin072, 2018-06-16:
@tobox Did you get it to do what you wanted? Had a look at your code and want to try somethings myself as well. 

@arendst can you give a small guide how to add custom (i2c) library's and keep it user configurable so with future sonoff update's I won't have to edit the main files? I want to implement the usage of a GridEye sensor (AMG88xx)

colinl, 2018-06-16:
@martin072 see also https://github.com/arendst/Sonoff-Tasmota/wiki/PID-Control-with-Sonoff-Devices.

closed 2018-08-15 by stale[bot]