arendst/Tasmota #1830: Unable to compile.   'RtcSettings' was not declared in this scope

cloggedDrain, 2018-02-08:
I'm trying to compile Tasmota for the first time and I'm receiving the following errors.  I have a feeling it's a simple rookie mistake, but I'm not sure how to troubleshoot this further.

```
H:\Arduino\arduino-1.8.5\portable\sketchbook\Sonoff\settings.ino: In function 'uint32_t GetRtcSettingsHash()':
 
settings:50: error: 'RtcSettings' was not declared in this scope
```
[Full error here](https://pastebin.com/Ar7YdvdG)

I've have been following this guide to set everything up:  https://github.com/arendst/Sonoff-Tasmota/wiki/Arduino-IDE

I've tried the development and master branch and receive the same errors when attempting to compile.

I'm using the following configuration:

```
Board: "Generic ESP8266 Module"
Flash Mode: "DOUT"
Flash Size: "1M (no SPIFFS)"
Debug port: "Disabled"
Debug Level: "None"
IwIP Variant: "v1.4 Prebuilt"
Reset Method: "nodemcu"
Crystal Frequency: "26 MHz"
Flash Frequency: "40MHz"
Upload Using: "Serial"
CPU Frequency: "80 MHz"
Upload Speed: "115200"
```

closed 2018-02-08 by cloggedDrain

TLCary, 2018-02-27:
Same problem.  Upgraded Mac OSX to 10.13 and then the trouble started.

Felias1701, 2018-04-28:
Same problem here as well (Win10).

andreasloe, 2018-07-31:
Unfortunately, this does not solve the problem for me. I am having exactly the same error message and I removed arduino completely from my Mac OS 10.13.6 in order to have a fresh start. All libraries were included, still 
`Users/whatever/Documents/Arduino/Sonoff/settings.ino: In function 'uint32_t GetRtcSettingsCrc()':
settings:75: error: 'RtcSettings' was not declared in this scope
   uint8_t *bytes = (uint8_t*)&RtcSettings;
 `
and I have the following versions
`Bibliothek esp-knx-ip-0.5.1 in Version 0.5.1 im Ordner: /Users/whatever/Documents/Arduino/libraries/esp-knx-ip-0.5.1  wird verwendet
Bibliothek EEPROM in Version 1.0 im Ordner: /Users/whatever/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/libraries/EEPROM  wird verwendet
Bibliothek ESP8266WiFi in Version 1.0 im Ordner: /Users/whatever/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/libraries/ESP8266WiFi  wird verwendet
Bibliothek ESP8266WebServer in Version 1.0 im Ordner: /Users/whatever/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/libraries/ESP8266WebServer  wird verwendet
Bibliothek Ticker in Version 1.0 im Ordner: /Users/whatever/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/libraries/Ticker  wird verwendet
Bibliothek ESP8266HTTPClient in Version 1.1 im Ordner: /Users/whatever/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/libraries/ESP8266HTTPClient  wird verwendet
Bibliothek ESP8266httpUpdate in Version 1.1 im Ordner: /Users/whatever/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/libraries/ESP8266httpUpdate  wird verwendet
Bibliothek ArduinoJson-5.11.2 in Version 5.11.2 im Ordner: /Users/whatever/Documents/Arduino/libraries/ArduinoJson-5.11.2  wird verwendet
Bibliothek DNSServer in Version 1.1.0 im Ordner: /Users/whatever/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/libraries/DNSServer  wird verwendet
Bibliothek ArduinoOTA in Version 1.0 im Ordner: /Users/whatever/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/libraries/ArduinoOTA  wird verwendet
Bibliothek ESP8266mDNS im Ordner: /Users/whatever/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/libraries/ESP8266mDNS (legacy) wird verwendet
Bibliothek PubSubClient-2.6.09 in Version 2.6 im Ordner: /Users/whatever/Documents/Arduino/libraries/PubSubClient-2.6.09  wird verwendet
Bibliothek TasmotaSerial-2.0.0 in Version 2.0.0 im Ordner: /Users/whatever/Documents/Arduino/libraries/TasmotaSerial-2.0.0  wird verwendet
Bibliothek C2Programmer-1.0.0 in Version 1.0.0 im Ordner: /Users/whatever/Documents/Arduino/libraries/C2Programmer-1.0.0  wird verwendet
`

andreasloe, 2018-07-31:
The error message 
`'RtcSettings' was not declared in this scope`
means that the variable is not defined. I found this variable in the following files
settings.h
settings.ino
sonoff.ino
support.ino
xdrv_03_energy.ino
xsns_01_counter.ino
support.ino
If I could know how this variable will be defined I could be able to locate the problem further. The closest seems to be settings.h with the lines
`struct RTCMEM {
  uint16_t      valid;                     // 000
  byte          oswatch_blocked_loop;      // 002
  uint8_t       ota_loader;                // 003
  unsigned long energy_kWhtoday;              // 004
  unsigned long energy_kWhtotal;              // 008
  unsigned long pulse_counter[MAX_COUNTERS];  // 00C
  power_t       power;                     // 01C
                                           // 020 next free location
} RtcSettings;`