arendst/Tasmota #1831: Avahi Problem

kskenyon, 2018-02-08:
I have several devices running on 5.11.1i.  I don't see any evidence that any of them are accessible through name-XXXX.local.  I can access several other things through Avahi/Bonjour/mDNS but not Tasmota devices.  I have tried several different Bonjour browsers (MacOS) and none of them detect Tasmota devices.  It used to work a couple of months ago.  I don't want to use fixed IP addresses and they often change with DHCP on my router.  I'm trying to keep things simple so having Avahi "just work" for Tasmota is what I'm looking for.  I don't know where to look to fix this.

// -- mDNS ----------------------------------------
#define USE_DISCOVERY                            // Enable mDNS for the following services (+8k code, +0.3k mem) - Disable by //
  #define WEBSERVER_ADVERTISE                    // Provide access to webserver by name <Hostname>.local/
  #define MQTT_HOST_DISCOVERY                    // Find MQTT host server (overrides MQTT_HOST if found)

The configuration web page also shows that these options are enabled and running.

Any help would be greatly appreciated!

Kevin

reloxx13, 2018-02-10:
heya,

check this point 1.

#1851

kskenyon, 2018-02-10:
Different problem.  Each Sonoff device web server is not showing up as broadcasting to mDNS like it used to. 

I have my MQTT server (mosquitto) set up to broadcast it’s address on mDNS but I still have it set in config.h. 

BTW I do use the config override feature but it gets overwritten once in a while by git pull. Last time I didn’t notice and did a mass update with the default config.h. Bad news ensued. 

Kevin

arendst, 2018-02-10:
Shouldn't the overwrite be stopped by .gitignore?
```
.pioenvs
.piolibdeps
.clang_complete
.gcc-flags.json
.vscode
sonoff/user_config_override.h
```

kskenyon, 2018-02-10:
It should but it didn’t the last time.  Was that recently added?

I keep a backup of user_config_override.h just in case and had to use that when the default from github overrode it on Thursday.  I didn’t notice when I did a build and update and now I have a bunch of fixing to do.  I should have verified the build before an OTA update.  I learned the hard way.  I do all my git pulls from the Sonoff-Tasmota directory so it should have worked.

I think the idea of having the default set for AP mode might not be a bad idea.  At least you can fix things then without having to reburn the devices.  Taking apart a Sonoff is a pain when it’s wired in.  That’s the whole point of OTA isn’t it?

BTW, Tasmota is the best thing since sliced bread.  Excellent work and I really appreciate it. It makes IoT fun.

Kevin

kskenyon, 2018-02-10:
I was doing a git stash before git pull since git wouldn't let me update without it.  I just realized that stash made git install the default user_config_overrride.h from GitHub. 

Duh.  How do I get around doing a git stash so git pull will work?  Not a Git wizard...yet.

Kevin