arendst/Tasmota #1832: SonofRF-Tasmota integration with homeassistant.

JeroenBruneel, 2018-02-08:
Hello Tasmota enthousiast.

I have bought a sonofRF and flashed the tasmota firmware. 

I also have an Sonof PIR, Keyfob and Doorsensor working on RF.

When i activate the detectors above I get the following logging in the tasmota console.

Sonoff PIR

19:12:40 MQT: tele/sonoffrf/RESULT = {"RfReceived":{"Sync":12470,"Low":420,"High":1220,"Data":"D2C71E","RfKey":"None"}}
19:13:22 MQT: tele/sonoffrf/RESULT = {"RfReceived":{"Sync":12380,"Low":420,"High":1220,"Data":"D2C71E","RfKey":"None"}}

Keyfob

ButtonA

19:18:01 MQT: tele/sonoffrf/RESULT = {"RfReceived":{"Sync":11080,"Low":350,"High":1080,"Data":"BC7DE4","RfKey":"None"}}
19:18:04 MQT: tele/sonoffrf/RESULT = {"RfReceived":{"Sync":11090,"Low":350,"High":1080,"Data":"BC7DE4","RfKey":"None"}}

ButtonB

19:18:36 MQT: tele/sonoffrf/RESULT = {"RfReceived":{"Sync":11130,"Low":360,"High":1060,"Data":"BC7DE1","RfKey":"None"}}
19:18:44 MQT: tele/sonoffrf/RESULT = {"RfReceived":{"Sync":11130,"Low":360,"High":1070,"Data":"BC7DE1","RfKey":"None"}}

ButtonC

19:19:25 MQT: tele/sonoffrf/RESULT = {"RfReceived":{"Sync":11090,"Low":350,"High":1080,"Data":"BC7DE8","RfKey":"None"}}
19:19:34 MQT: tele/sonoffrf/RESULT = {"RfReceived":{"Sync":11090,"Low":350,"High":1080,"Data":"BC7DE8","RfKey":"None"}}

ButtonD

19:20:03 MQT: tele/sonoffrf/RESULT = {"RfReceived":{"Sync":11130,"Low":360,"High":1070,"Data":"BC7DE2","RfKey":"None"}}
19:20:05 MQT: tele/sonoffrf/RESULT = {"RfReceived":{"Sync":11120,"Low":360,"High":1070,"Data":"BC7DE2","RfKey":"None"}}

PIR

19:21:08 MQT: tele/sonoffrf/RESULT = {"RfReceived":{"Sync":14190,"Low":510,"High":1470,"Data":"946A21","RfKey":"None"}}
19:21:23 MQT: tele/sonoffrf/RESULT = {"RfReceived":{"Sync":13820,"Low":490,"High":1430,"Data":"946A21","RfKey":"None"}}

I see that Sync,Low,High feedback differs but that the Data is consistent per detector.

Can anyone point me in the right direction to integrating these sensors in Home assistant ?

I am trying to build a basic alarm system with these.

tyjtyj, 2018-02-08:
off topic. Can you share with me how u flash sonoff rf bridge ? i keep getting md5 error flashing esptool

back to topic

u can use https://home-assistant.io/components/binary_sensor.mqtt/

Here is possible working solution...

binary_sensor:
  - platform: mqtt
    name: "PIR Sensor"
    state_topic: "tele/sonoffrf/RESULT"
#based on high value
    payload_on: "1470"
    payload_off: "1430"
    availability_topic: "tele/sonoffrf/lwt"
    payload_available: "online"
    payload_not_available: "offline"
    qos: 0
    device_class: opening
    value_template: '{{ value_json.RfReceived.High }}'

Sigma95, 2018-03-07:
Jeroen,

How did you have proceeded ? I am in the very same situation.

tyjtyj, 2018-03-07:
Here is what i am using 

```
- alias: 'AutoMQTTRF433_Ikea toggle'
  trigger:
    - platform: mqtt
      topic: 'tele/sonoff/rf/RESULT'
  condition:
    - condition: template
      value_template: '{{ trigger.payload_json.RfReceived.Data == "123456" }}' 
  action:
    - service: switch.toggle
      entity_id: switch.ikea_light
```

Sigma95, 2018-03-08:
thanks tyjtyj,

Do you know if it's possible to do a similar solution in OpenHab2?

abaksa1, 2018-03-16:
Thanks tyjtyj, confirmed it is working