arendst/Tasmota #1838: Feature Request...

TLCary, 2018-02-09:
I'm trying to make a Garage Door Opener with the wonderfully versatile Tasmota.
I have limit switches for Door Open and Door Closed created as Switch3 and Switch4. I can pull the "Power3" or "Power4" out of the cmd/<custom switch topic>/Power<x> = <StateText> MQTT response to tell if the door is opened or closed.  I can make it work...
But it would be elegant if a switch state would be reported as a:  stat/<custom switch topic>/  switch<x> = <StateText>.  Having a switchtopic<x> for each switch instead of one global might also be worth considering.
Lastly, I have to momentarily, (Like 500 milliseconds), close the relay and open it to make the pulse to simulate pushing the wall button to get the garage door opener to activate.  Is there a way to limit the relay on time?  If I generate the OFF command through MQTT there will be a great deal of variance in how long the ON phase will be. 
Yeah, I as a lot...  Worse case scenario, I get a NO.  Thanks for all your hard, and well done, work.

Frogmore42, 2018-02-09:
Wiki Main page "PulseTime" command for the 500ms question

ascillato, 2018-06-02:
Hi @TLCary 

If your issue was solved, please close it. Thanks :+1:

closed 2018-06-02 by TLCary