arendst/Tasmota #1839: Problems with Sonoff T1 Touch UK (2 buttons)

mattpurnell, 2018-02-09:
Hi all.

Firstly - I have this web issue on all my Tasmota devices; including the other single-button T1 switches I have that work fine;. The first problem - the websites all show like this, and Chrome, Firefox and IE:

[image]

I don't know what the version is for the T1, but it is brand new. I pulled the code down only a few weeks ago. I am using Arduino1.8.5 to flash the devices. My flashed devices all work fine in terms of connectivity and function, but the web server displays the same on all of them, even a stock standard ESP8266.

I have tried to find other people with this issue, but I haven't found anything other than a guy for whom the webserver was not working at all. He had a fix by editing platformio.ini - but I am using Arduino 1.8.5. 

The fact that I am unable to access those web options will more-than-likely explain away my second issue, as I am unable to perform step 4 of the "initial Configuration": **4. Select the correct Sonoff module type from the configuration frontend**

The second light switch on the board does nothing. The first light switch works and sends an MQTT message that implies there is no config for a second switch (stat/lightlounge/RESULT` {"POWER":"ON"}). I am hoping that selecting the correct model from the web server will fix that.

Anyone that can point me to the right fix will be a hero to me!

Cheers
Matt

Frogmore42, 2018-02-09:
It sounds like it could be bad signal for Wi-Fi.  Are there metal boxes for the wiring?  How far away is the router/access point?

If MQTT is working you can also use it to configure.

I would also suggest trying the latest version since there has been some back and forth on the web server code.

Frogmore42, 2018-02-09:
Issue 1826 has some suggestions for using the older core for Arduino, which seems to work better.

mattpurnell, 2018-02-09:
Thanks for the feedback - I can't blame wifi, it was on the table about 3m from the router. I'll look into 1826 - cheers!

tobox, 2018-02-09:
I think the problem is related to chunked transfers for the webserver. Try to flash a version newer than 5.11.1f, that should solve your problem.

mattpurnell, 2018-02-09:
Thanks @tobox I'll give that a go.

@Frogmore42 - can I configure the module type with mqtt?

Frogmore42, 2018-02-09:
Yes, I believe you can.  There is a list of commands on the wiki.

mattpurnell, 2018-02-10:
@Frogmore42  I had looked before asking, and I have looked again - I couldn't see that option I'm afraid. I looked in MQTT Features and MQTT Overview. All good though, I am going to try other versions

Frogmore42, 2018-02-10:
Module command will do it.

Frogmore42, 2018-02-10:
https://github.com/arendst/Sonoff-Tasmota/wiki/Commands

mattpurnell, 2018-02-10:
@Frogmore42 thank you for showing me that... and, good god.. what a huge array of features and commands this project supports!? Amazing they've packed so much into such a tiny device..

mattpurnell, 2018-02-10:
PS I am trying to provide an update, but the update totally broke my arduino project and I couldn't get it going again. So I have swapped over to Atom+PlatformIO - and it isn't uploading correctly either. The serial monitor reads correct datestamps but nothing legible. I am starting to think my computer has something wrong with it - I am ready to rebuild my laptop..
```
00:07:56 RSL: ��������������������������������������������������������������������������������������������������� = �����������������������
�������������
```

Frogmore42, 2018-02-11:
sounds like you serial connection is getting confused, but if you have some text it appears that you were successful in flashing it.  See if you can find the it's IP address from your router (many of them will show you the list of devices that have gotten addresses).  Then try going to the web page and configure from there.

mattpurnell, 2018-02-11:
Thanks @Frogmore42  but no, there is no successful connection to the wireless network. If DHCP succeeded I would see the lease in my router.

Frogmore42, 2018-02-11:
Are you using DHCP or trying to set a static IP address in the config?
I always use DHCP, since my router assigns IP addresses based on MAC address (and also allows for manual static configuration) it is pretty easy to find my devices and I have a OneNote with all of them listed, which makes it even easier.

I was never able to get a Sonoff device to partially flash, it either did it or failed.  It sounds like you might have an issue with getting it flashed correctly.  So then you have to look at how you are powering it and what USB-Serial converter you are using and what is it connected to.

mattpurnell, 2018-02-12:
I have tried DHCP and static.. neither succeed. I track mine with DNS names, and apparently they boot into action much faster if static.

Yes it's so weirdly consistent, I have just tried another clean build in Arduino IDE - exactly the same thing, Frustration is through the roof!

Frogmore42, 2018-02-13:
What are you using for a router/access point?  Some people have had issues with particular ones.

How saturated is your environment for Wi-Fi (how many neighbors have networks)?

What channels are they on?

It is also possible your Sonoff is defective.  You might get a NodeMCU device, like the wemos d1 mini to get some experience without the possibility of electrocuting yourself.

mattpurnell, 2018-02-14:
Thanks for all your help BTW - but I don't think it was an issue with the wireless or the device, the behaviour was too consistent and too defective - reporting an IP of 255.255.255.255!?

I have a number of ESP8266-based and none of them were flashing right.. so what I have eventually done is rebuild my windows laptop. I hope to have an update soon..

tobox, 2018-02-14:
>  reporting an IP of 255.255.255.25 !?

I once had the same issue, the saw that the Sonoff was connected to the access point, but I was unable to ping it. I saw IP=255.255.255.255 using the serial log, and my quick and dirty solution was to rebuild the firmware with a difference value for CFG_HOLDER (that reverts the settings to default).

mattpurnell, 2018-02-14:
Thanks @tobox I'll try that

mattpurnell, 2018-02-22:
Hi @tobox  that got my devices flashing again - great.. one problem down. Thanks so much.

But my webserver is still serving up the same old garbage - even after a complete laptop rebuild.

Every page has loads of errors, but I can get as far as seeing the 'Module' button I really want to use. But it doesn't work :( 

Where can the error lie? I am beyond frustrated now, I am just purely baffled

[image]

mattpurnell, 2018-02-22:
@Frogmore42 thanks for the MQTT suggestion; after changing the CFG_HOLDER value I got it flashing again.. and even though I can get very little sense out of the webserver, the MQTT configuration option was very easy and worked a treat.

Thanks everyone for your assistance!

closed 2018-02-22 by mattpurnell

lazers378, 2018-08-16:
I have version 5.14.0 and having similar problem. But the problem is random. Sometimes web server responds just fine and some times home page shows up and when i click on configuration it shows such garbage page or somtimes on timer page is garbage. 
Any solutions ?

Frogmore42, 2018-08-16:
This issue is closed.  Your issue is likely not the same.  If you believe you have an actual issue, please follow the template and provide all of the requested information in a new issue.