arendst/Tasmota #1840: Wifi qualitiy (RSSI) dropped to unusable after update to 5.11.1

maciboy, 2018-02-09:
Hi,
I’m using a Basic module and was fine with it the last 4-5 weeks when running v5.9.
Now I updated to v5.11.1 and my wifi signal strength dropped significantly.
Before at the place where the Sonoff lives it has about 20% signal strength, now it blinks and says that AP was not found.
Moving the Sonoff closer to the router helps, but signal strength is still worse than before.
Resetting the module did not help…

Can this be an effect of the update, since that’s the only thing that changed?
Thanks for any advices!
(Flashed with SonOTA)

curzon01, 2018-02-10:
What's happening with the signal strength when you downgrade to v5.9.1 again? Will the probem stay?

maciboy, 2018-02-10:
I now tried signal strength with the new 5.12 release and also downgraded to 5.9.1: Wifi performance stays as bad as it was 😢 
How can it be that wifi performance degrades at a moment's notice?

ErikAndren, 2018-02-10:
If you have an android phone you could check how many transmitters there are on the different wifi channels by using an app like "Wifi Analyzer". Maybe a neighbour installed a new access point on the same channel?

maciboy, 2018-02-10:
[image]
I could not recognize any change there. The Networks my Max receives at the spot are the same as before and mine is the only one on channel 1 - with quite good reception…

curzon01, 2018-02-11:
You have changed something:

Did you move the basic module (rotate, changed position) or is it in exact the same position?
I don't thing so, because you wrote you moved it to a shorter position for test  - and than back - but in abolsutely same rotation and position?

Signal strength for the ESP are also depending on rotation of the ESP antenna (in relation to the antenna of your router). Try this one.