arendst/Tasmota #1841: How to Flash the RF Bridge.

erew123, 2018-02-09:
Someone asked me on here a more detailed explanation of how to do this, so I thought I would post (as best as I can recall) how to do it with a Windows PC.

**TASMOTA Source Code zip file and extract it (I used 5.11.1b I think)**
https://github.com/arendst/Sonoff-Tasmota/releases

**Arduino IDE (Im on 1.86)**
https://www.arduino.cc/en/Main/Software

1) Install/Extract the Arduino IDE software.

2) In the folder where you installed Arduino IDE to, make a new folder called **portable**

3) Open the Arduino IDE sofware . Go to **File** > **Preferences** > **Additional Board Manager** and add this URL:
http://arduino.esp8266.com/versions/2.3.0/package_esp8266com_index.json

4) Go to **Sketch** > **Include Library** > **Manage Libraries**. Search for and install **ArduinoJson** and **PubSubClient** 

5) In the “**portable**” folder you created in step 2, go into that folder > **libraries** and copy "**IRremoteESP8266**" and "**PubSubClient**" into your **Tasmota\lib** folder.

6) Now in your **Tasmota\lib\PubSubClient\SRC** folder, open **PubSubClient.h** with a text editor (Notepad) and change “**MQTT_MAX_PACKET_SIZE**” to “**512**”. Save and exit. 

7) Back in Arduino IDE go to **File** > **Open**. Browse to the the folder where you extracted the TASMOTA firmware and open **sonoff-sonoff.ino**.

8) Find the **user_config.h** tab at the top right of the Arduino IDE interface (on a drop down arrow) and within that file, find the sections where you can:

Enter your Wifi name and password
Enter your MQTT broker IP address, port, username and password

9) On the **Tools** menu, ensure you set these settings:
[https://i.imgur.com/sTOctw1.png](url)

**Board:** Generic ESP8266 Module
**Flashmode:** DOUT
**Freq:** 40
**CPU:** 80
**Flash:** 1M (64K SPIFFS)
**Upload speed:** 115200
**Port:** Com (THIS WILL NOT BE SELECTABLE UNTIL YOU CONNECT YOUR USB SERIAL ADAPTER.... SO SELECT THIS AFTER YOU HAVE PLUGGED IN THE BRIDGE CORRECTLY WHILE HOLDING DOWN THE PAIRING BUTTON)

10) Ensure the switch on your RF Bridge is switched of the OFF position (change it back to ON position **AFTER** flashing your Bridge).

11) Connect up your USB serial adapter to the correct pins on the bridge. Hold DOWN the pairing button while connecting the USB serial adapter to your computer (the bridge should now go into the correct mode where you can flash it). (Pin out etc on the below image)
[https://i.imgur.com/djhr2Jg.jpg](url)

12) In the Arduino IDE interface, you can now select the correct PORT on the **Tools** menu.

13) At the top right of the Arduio IDE software, click on the right arrow (UPLOAD) to upload the firmware. This should take about 2-3 minutes as it compiles the firmware then uploads it.

When compete, disconnect the cables etc, switch the switch back to the **OFF** position and it should now be working as you expect.

reloxx13, 2018-02-09:
https://github.com/arendst/Sonoff-Tasmota/wiki/How-to-Flash-the-RF-Bridge

erew123, 2018-02-09:
I edited the image on here and on the WIKI page you created, along with changing the SPIFFS setting above!

Hope this helps people in future!

ptrooms, 2018-04-05:
I don't understand/get it..... what't the purpose of the "portable" folder (which I've added per your instruction to /home/xxx/sketchbook/libraries/)
Installed the named ibrarues where/how to I (physically) locate the "IRremoteESP8266" and "PubSubClient" folders for copy to Tasmota ?

FYI: 
I've installed de Arduino IDE 1.85  using Ubuntu 16.04 Software which created:
an empty folder: /home/xxx/sketchbook/libraries/
I maintain each .ino sources as folder in:  /home/xxx/Arduino/

TIA Peter

erew123, 2018-04-06:
Hi Peter

I cant give you a 100% answer on this..... however, when the Sketch software builds the firmware, it requires certain library files.... one of those being the IRremoteESP8266 library (as well as some others)

I used a Windows version of Sketch, Im not sure if that makes any difference or not.

What I can tell you is that IF I didnt move an additional copy from the default location where these files are installed, then when I told Sketch to build and copy the firmware to any Sonoff device, it would fail, claiming that it couldnt find the IRremoteESP8266 library (or some other libraries).

After hunting the internet for using Sketch with Sonoff's, I found that pretty much all the instructions indicate you need to copy the files to this additional location (again, Im not sure if this is a Windows OS thing or not). 

For example, there were a couple of videos on youtube, that showed the instructions for using Sketch with Tasmota/Sonoff's, how to set it up etc.... What I found in many of these videos were the missed this step in their video and in the description box..... though hunting through the comments, I found a few people suggesting to copy the files to the location I mentioned, and all of a sudden, Sketch could find the files it needed, it compiled the firmware without complaint and installed the firmware on the Sonoff.

So in answer to your question... I dont know if its a Windows OS only issue, a Sketch only issue, or perhaps a bug for the version of Sketch that I used..... but moving the files there made it work.

I guess if you find it just works on Ubuntu without moving/copying the files to this location, leave a note back here an I will update the original post.

Thanks

ptrooms, 2018-04-06:
Solved, thanks in massive return.Your excellent instructions helped me greatly.

I've found my (ownlearning by doing) solution, was able to compile and flash. 
In summary, I placed the required _Tasmota libraries_ in my personalized default  .../Arduino/libraries (sketch location) folder.

For Ubuntu users, you could consider to add an instruction remark for step 4 & 5:  place/copy the required Tasmota function libraries into the  **libraries** folder of the Arduino **sketch folder**. The Sketch book location of  the Arduino IDE can be found my looking up preferences *Menu File >>  Preferences".  

### Below a few more detailing words for the archive,
Initially a (new) Arduino IDE user will find a /home/username/sketchbook/**libraries** folder that hold personalized libraries folder.  I my case, I did change this folder name and location my changing preferences.

My Arduino/Sketch installation on Ubuntu 16.04  was done via regular  Software-Install taken from the Ubuntu store. The installation creates the Arduino sketch folder, according to preferences (Menu >> Preferences >> Sketchbook_location)  for the user in /home/username/... (equivalent to "c:\users\username\My Documents" on Windows).

Note: Aside a "/home/username//home/**sketchbook**/"with a subfolder /home/username/sketchbook/**libraries** was created. I believe this was _meant_ as the original "sketch"folder which I , in my ignorance, via settings have changed to  "/home/username/**Arduino** . However, this folder, for me is now superfluous.

When starting the IDE one have/defines a standard "source"  location for coding, in my case "/home/username/**Arduino**" which (will) contain an empty folder "**libraries**" initially having only a "readme.txt"  containing "For information on installing libraries, see: [http://www.arduino.cc/en/Guide/Libraries](http://www.arduino.cc/en/Guide/Libraries)"

### What I've done: 
- [] cloned Tasmota source got into a personalised folder
- unzipped into code-folder /home/username/**code-tasmota**/sonoff/sonoff.ino
- [ ] Ignored step 2, did not create a "portable" (sub)folder.
- [ ] Following your step 4. retrieved the following libraries,  using the Arduino IDE Library manager into my private Arduino sketch library folder (/home/username/**Arduino/libraries/**): 
- PubSubClient
- ArduinoJson
- [ ]  Changed step 5 by manually copying the following Tasmota library subfolders (in ...Tasmota/**lib/**) towards my private Arduino sketch library folder (/home/username/**Arduino/libraries/**): 
- Adafruit_SGP30-1.0.0.13
- IRremoteESP8266
- NeoPixelBus-2.2.9
- TasmotaSerial-1.2.0
In the Arduino IDE menu "Sketch>>IncludeLibraries>>"  one can see that the contents is nicely picked up by de IDE.
- [ ] Changed step 6  (edit  MQTT packetsize to 512 bytes)
- by editing my personalised /home/username/Arduino/**libraries/PubSubClient**/src/PubSubClient.h

Thereafter, (your step 7) opening  /home/username/code-tasmota/sonoff/**sonoff.ino**  and following  result into the SonOff upload via USB serial connection  , was a walk in the park.

Great work !!!!

PauloPC, 2018-05-24:
Sonoff RF Bridge compiling problem - exec: "make": executable file not found in %PATH%

I'm trying to flash it with TASMOTA 5.14, but I tried with older versions too and got same message.

May board config is:
[image]

I have no idia where to find e MAKE Path.

erew123, 2018-05-24:
Closest I can find to an answer is this 

https://github.com/arendst/Sonoff-Tasmota/issues/2256

ascillato, 2018-06-13:
Hi

If your issue was solved, please close it. Thanks :+1:

closed 2018-06-14 by erew123