arendst/Tasmota #1843: Help me please

ZoltXs, 2018-02-09:
Hello, I'm new and I need help, someone could tell me if there is any tutorial like connect sonoff to mqtt mosquito server on Mac OS, I installed tasmota firmware but I do not know how to connect mqtt mosquitto to sonoff switch basic.

reloxx13, 2018-02-09:
heya,

maybe this will help you: https://github.com/arendst/Sonoff-Tasmota/wiki/MQTT-Features

ZoltXs, 2018-02-09:
thanks but I do not understand it 😢

TLCary, 2018-02-09:
In a nutshell:  Install Tasmota on your Sonoff.   Give the button on the Sonoff 4 short presses to put it into Wifi Access Point Mode.  On your Mac instead of connecting to your normal WiFi access point, connect to the Sonoff device.  Point a web browser to 192.168.4.1  (which is a web server on the Sonoff device). From the menu Configure/Configure MQTT/  enter the address of your Mac in Hosts...  Push the button again and put your Mac back on your normal Wifi.

ZoltXs, 2018-02-09:
and the user and password I have not configured any in mqtt server are there any defaults?

TLCary, 2018-02-09:
No. If MQTT server isn’t configured to require login you don’t have to provide any at all, or use any user/password you like. Doesn’t matter.

closed 2018-02-13 by ZoltXs