arendst/Tasmota #1845: sonoff-minimal

kskenyon, 2018-02-10:
I use Atom with platformio to build tasmota.  Is there a way to select more than one build without building every language?  I just want sonoff-minimal and sonoff for example to be built without having to edit platformio.ini.  It looks like you can only select one build at a time or all of them.

arendst, 2018-02-10:
Ask platformio forum as it's a platformio issue.

reloxx13, 2018-02-10:
heya,,

yep, you can comma seperate it like this:

`env_default = sonoff-DE, sonoff-minimal`

arendst, 2018-02-10:
Why can't life always be so simple...

reloxx13, 2018-02-10:
haha yep, but it took some time to find this information :D

their is also a way to solder in a 4mb flash to the sonoff devices, no need for minimal than 👍 

just ordered some flash from china :D

kskenyon, 2018-02-10:
Thank you.  Works great.

Frogmore42, 2018-02-10:
You will still need minimal, as Tasmota is not configured to take advantage of more that 1MB of flash.

Jason2866, 2018-02-11:
One question,
i successfully flashed my devices from 5.11.1 developer to 5.12 master via the webconsole.
No need to use minimal Version. Do you have an enhanced Version, which needs more memory?

Frogmore42, 2018-02-12:
I don't, but there were some interim versions that did not fit and it probably won't be long before stuff will need to be not included to fit within the constraints of 1MB.  I don't believe it is a real problem, since it is possible to have multiple binaries that have different options enabled.  It is nice to have a single binary that works across all of my devices, but I don't believe that is a requirement.