arendst/Tasmota #1846: Version 5.12.0 causes Watchdog Timeout Reset on Sonoff-POW

DeanCording, 2018-02-10:
I have a number of Sonoff devices including Basic, TH, Dual and POW which I recently upgraded to 5.12.0

All devices upgraded successfully except for the two POW devices which get stuck in a watchdog timeout loop.

```
00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_922A7F, GroupTopic sonoffs) Version 5.12.0
00:00:00 WIF: Connecting to AP1 Home in mode 11N as sonoff-2687...
00:00:05 WIF: Connected

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v4ceabea9
~ld

00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_922A7F, GroupTopic sonoffs) Version 5.12.0
00:00:00 WIF: Connecting to AP1 Home in mode 11N as sonoff-2687...
14:01:17 WIF: Connected

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v4ceabea9
~ld

00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_922A7F, GroupTopic sonoffs) Version 5.12.0
00:00:00 WIF: Connecting to AP1 Home in mode 11N as sonoff-2687...
00:00:04 WIF: Connected

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v4ceabea9
~ld

00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_922A7F, GroupTopic sonoffs) Version 5.12.0
00:00:00 WIF: Connecting to AP1 Home in mode 11N as sonoff-2687...
00:00:04 WIF: Connected
```

I have tried disabling all sensors and other options except MQTT.  I have also change the CFG_HOLDER to reset the configuration.  Nothing has worked.

All my other Sonoff Basic, TH, and Dual devices work fine.

arendst, 2018-02-10:
Pls provide output of command ``status 0``.

DeanCording, 2018-02-10:
It doesn't stay up long enough to connect to it.  The reboots happen about every 15 seconds.  I notice that it occasionally can sync with NTP before rebooting.

LwIP  variant is  V2 prebuilt (MSS=536)

Changing to LwIP variant v!.4 prebuilt works fine.  Interesting that it only affects the POW devices and even when they are configured as Basic devices.

Thanks

closed 2018-02-11 by DeanCording

davidelang, 2018-02-11:
if you can make your image log over the network you may get something, otherwise 
you have trouble because you need an opto-isolated serial connection to avoid 
frying your computer

chimeranzl, 2018-02-28:
Exactly the same issue here with Sonoff POW.  Boots up, connects to Wifi, says its connected, get 1 ping response, then wdt reset. EG:

```
00:00:00 Project sonoff Laundry (Topic laundry, Fallback DVES_B49EBB, GroupTopic sonoffs) Version 5.12.0
00:00:00 WIF: Connecting to AP1 iotbam in mode 11N as laundrysonoff...
00:00:03 WIF: Connected

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v4ceabea9
~ld

00:00:00 Project sonoff Laundry (Topic laundry, Fallback DVES_B49EBB, GroupTopic sonoffs) Version 5.12.0
00:00:00 WIF: Connecting to AP1 iotbam in mode 11N as laundrysonoff...
00:00:03 WIF: Connected

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v4ceabea9
~ld

```