arendst/Tasmota #1848: AiLight w/ Hue Bridge emulation, no color on Echo

zerog2k, 2018-02-10:
I have an AiLight running latest (5.11.1j), running Hue Bridge emulation, and being controlled by Alexa/Echo.
I can turn it on and off, but there are a few issues with dimming/color:
1. it only shows a brightness dimmer, and although the AiLight is RGBW, changing the brightness dimmer through alexa app, only seems to change the RGB values together (not W). Is there a way to control this "mapping" between the hue bridge "slider" range to RGBW values on the light?
2. the alexa shows that the "color capabilities" are "none" - so it appears as a white dimmable light only. I understand that hue should actually support colored bulbs, so is there a way to emulate this on tasmota?

altelch, 2018-02-10:
Hi,

the Echo only support a very basic HUE support. You can controll on/off 
and brightness. To get color support you need to use the HUE app but 
Tasmota isn't a real HUE Bridge with cloud connection so this can't work.

Greetings,

      Altelch.

zerog2k, 2018-02-10:
Ok interesting. Strange that it even mentions color control ability in the app, and I also read some places where the supposedly the echo can control hue color natively (as of a year or so ago), but I don’t have a real bridge & bulb to test and verify.
On issue 1, I’ll look more later but I noticed the hue functions in tasmota to set hsb implemented on the lights only sets rgb values and not the W channel.  Seems like a slight bug in implementation.  I’ll try my hand at a patch.

zerog2k, 2018-02-10:
For ref, here is a nice blog post on hsi to rgbw, with a geeky jaunt into color theory ;)
http://blog.saikoled.com/post/44677718712/how-to-convert-from-hsi-to-rgb-white

zerog2k, 2018-02-10:
More ref: (notes to self)
https://developers.meethue.com/documentation/core-concepts

zerog2k, 2018-02-12:
Actually, I'm not sure if this is covered in another issue, but even the tasmota web ui has a single brightness? ("dark" to "light") slider. Moving this only seems to keep some "memory" of the color mix, and doesnt give a consistent control. I'm wondering if there are plans to put in more controls (like HSV) on the UI?