arendst/Tasmota #1849: Home Assistant restart state not recognized

ap88, 2018-02-10:
First of all, great work!
I use the new mqtt discovery feature in HA and it works fine. But there is an issue when HA restarts, it can not get the retained state from the mqtt server (in this case the state of a POW switch). I discovered the bug in "xdrv_07_home_assistant.ino" where the config string is declared. By changing the following line 110 it works as expected:

from
`GetTopic_P(state_topic, STAT, Settings.mqtt_topic, S_RSLT_POWER);`

to
`GetTopic_P(state_topic, STAT, Settings.mqtt_topic, S_RSLT_RESULT);`

arendst, 2018-02-10:
What version of Tasmota are you using? The latest released versions (5.12.0 and 5.12.0a) both have line 110 set to:
```
 GetTopic_P(state_topic, STAT, Settings.mqtt_topic, S_RSLT_RESULT);
```

ap88, 2018-02-10:
Oh, my mistake! I swapped from/to. My version is 5.12.0a

arendst, 2018-02-10:
Please check SetOption4 to be set to 0.

arendst, 2018-02-10:
In that case it is as I expected.

As you are wise enough to disable retained messages you should execute the ``state`` command to feed Hass with the latest device status. Idealy this would be a script executed by Hass when it restarts. My knowledge of Hass is inadequate to provide this script but from it's documentation it must be possible.

ap88, 2018-02-10:
In a manual configuration for the switch as in your HA example, the string for the state_topic looks like this:

`state_topic: "stat/sonoff1/POWER"`

This is also the retained message which is set by PowerRetain. In the new mqtt config string the state_topic looks like this:

`ha/switch/sonoffp1/config = {"name":"Stromzähler","command_topic":"cmnd/sonoffp1/POWER","state_topic":"stat/sonoffp1/RESULT","value_template":"{{value_json.POWER}}","payload_off":"OFF","payload_on":"ON","availability_topic":"tele/sonoffp1/LWT","payload_available":"online","payload_not_available":"offline"} (beibehalten)
`

What exactly does SetOption4?

arendst, 2018-02-10:
SetOptions4 switches between response RESULT or Command name. For Hass I prefer the response RESULT as the Hass JSON decoding then better works.

ap88, 2018-02-10:
Ok, so the discrepancy is then the retain message which is sent when PowerRetain is ON. This is always stat/sonoff/POWER and not stat/sonoffp1/RESULT.

arendst, 2018-02-10:
Yes. That's why I prefer the JSON messages which are (with SetOption4 set) advertised on RESULT which in turn is easier for Hass to recognize.

ap88, 2018-02-10:
I commit, that HA gets the state in JSON correct but only, if a message as in the config string "state_topic" would be published retained. Publish the retained message depending on the SetOption4 would solve this and no further configuration would be needed.

arendst, 2018-02-10:
But as I hate retained messages I would prefer someone to come up with Hass restart script executing command ``/cmnd/sonoffs/state`` where sonoffs would be the grouptopic of all devices to let Hass receive state messages from all of them at once.

bensuffolk, 2018-02-14:
This is my restart automation to check status:

```
- action:
  - data:
      topic: tasmota/cmnd/t1/office/POWER
    service: mqtt.publish
  - data:
      topic: tasmota/cmnd/t1/conservatory/POWER1
    service: mqtt.publish
  - data:
      topic: tasmota/cmnd/t1/conservatory/POWER2
    service: mqtt.publish
  - data:
      topic: tasmota/cmnd/b1/bedroom/front/POWER
    service: mqtt.publish
  - data:
      topic: tasmota/cmnd/sonoffs/POWER
    service: mqtt.publish
  alias: Query Status on startup
  condition: []
  id: '1515866976409'
  trigger:
  - event: start
    platform: homeassistant
```

Its in an odd formatted order because that how HA seems to do it is you use the automation builder in the configuration section.

Also notice that I have had to include some of the sonoffs as specific devices as my B1 and T1's don;t respond to the generic topic for some reason (have not investigated why)

Ben

ap88, 2018-02-18:
For me the following config works for all Sonoffs also for T1's:

```
- action:
  - data:
      topic: cmnd/sonoffs/POWER
    service: mqtt.publish
  - data:
      topic: cmnd/sonoffs/POWER2
    service: mqtt.publish
  - data:
      topic: cmnd/sonoffs/POWER3
    service: mqtt.publish
  alias: Systemstart
  condition: []
  id: '1518951376931'
  trigger:
  - event: start
    platform: homeassistant
```

For all my devices GroupTopic is set to sonoffs.

Koenkk, 2018-04-30:
I find it a bit confusing that when using home assistant discovery, retaining power state **does not** work out of the box and requires extra configuration. The reason why you want to use home assistant MQTT discovery after all is because you don't want to configure anything...

This is also inconsistent with the documentation of Tasmota

Tasmota **with** home assistant discovery:
```json
{
    "command_topic":"stat/sonoff/POWER",
    "state_topic":"stat/sonoff/RESULT"
    ...
}
```

[Tasmota **without** home assistant discovery](https://github.com/arendst/Sonoff-Tasmota/wiki/Home-Assistant#switch):
```json
{
    "state_topic": "stat/sonoff/POWER",
    "command_topic": "cmnd/sonoff/POWER"
    ...
}
```

coldfire84, 2018-06-16:
> I find it a bit confusing that when using home assistant discovery, retaining power state does not work out of the box and requires extra configuration.

@Koenkk, what is specifically needed here? I have PowerRetain enabled on my Sonoff-Tasmota flashed (v5.14.0)  device, but Home Assistant (using MQTT discovery) after a restart does not show correct state of Sonoff. 

Using Mosquitto for MQTT with persistence enabled, so with the Sonoff set with PowerRetain to On I am not sure what else I am missing.

Koenkk, 2018-06-16:
@coldfire84  I'm currently using @ap88 solution to request the device status on home assistant startup.

coldfire84, 2018-06-17:
@Koenkk, thanks for clarifying - using configuration.yaml power states are correct after HA restart alongside:
- Home Assistant MQTT discovery being enabled
- Sonoff PowerRetain enabled

`automation:
  - alias: "Home Assistant Query Sonoff State on Startup"
    trigger:
      - platform: homeassistant
        event: start
    action:
      - service: mqtt.publish
        data:
         topic: cmnd/sonoffs/POWER
      - service: mqtt.publish
        data:
         topic: cmnd/sonoffs/POWER2
      - service: mqtt.publish
        data:
         topic: cmnd/sonoffs/POWER3`

ascillato2, 2018-07-10:
Hi,

Thanks a lot for sharing! Very useful. I'm using this now and works great :+1: 

Closing this issue. Please reopen if needed. Thanks

closed 2018-07-10 by ascillato2

crus1313, 2018-10-27:
> For me the following config works for all Sonoffs also for T1's:
> 
> ```
> - action:
>   - data:
>       topic: cmnd/sonoffs/POWER
>     service: mqtt.publish
>   - data:
>       topic: cmnd/sonoffs/POWER2
>     service: mqtt.publish
>   - data:
>       topic: cmnd/sonoffs/POWER3
>     service: mqtt.publish
>   alias: Systemstart
>   condition: []
>   id: '1518951376931'
>   trigger:
>   - event: start
>     platform: homeassistant
> ```
> For all my devices GroupTopic is set to sonoffs.
> This works for my Basic, Pow, Dual, TH10, T1 (EU, GB).

That 'id' is something specific for you or a general use ID? From where did you get it?

ascillato, 2018-10-27:
Sonoffs fallback topic is in all Tasmotas by default. You can change that if you want. See my_user_config.h file

ap88, 2018-10-27:
Ignore this id tag. It's just generated by the automation editor in home assistant.

crus1313, 2018-10-27:
> Ignore this id tag. It's just generated by the automation editor in home assistant.

What do you mean 'ignore'? Like delete it from there, put the same ID as above, will be dynamically generated in the above sketch?

ap88, 2018-10-27:
Yes you can delete it.